smoltcp-rs/smoltcp #7: How to actively open a TCP connection?

little-dude, 2017-03-04:
After looking at the doc and at the code, I could not find out how to actively open a TCP connection. Is this implemented? If so, how can it be done?

little-dude, 2017-03-04:
From the doc, it seems that it's there though:

> A TCP socket may passively listen for connections *or actively connect to another endpoint*.

whitequark, 2017-03-05:
> From the doc, it seems that it's there though:
> 
> A TCP socket may passively listen for connections or actively connect to another endpoint.

It's kinda not. Almost all of the machinery is there but I never implemented the front-end method for connecting somewhere.

whitequark, 2017-03-05:
Fixed in 1a32b98b.

closed 2017-03-05 by whitequark

whitequark, 2017-03-05:
@little-dude Please see the README for an example of using the TCP client feature.