smoltcp-rs/smoltcp #10: If mac dest is not broadcast/multicast or configured on the device, the frame should be ignored

little-dude, 2017-03-06:
Currently, smoltcp does not check the destination MAC address.

Here is an example (it reuses `examples/utils.rs`), where we create a device with hardware address `00:00:00:00:00:01` and ip `10.0.0.1`:

```rust
#[macro_use]
extern crate log;
extern crate env_logger;
extern crate getopts;
extern crate smoltcp;

mod utils;

use std::str;
use smoltcp::phy::RawSocket;
use std::time::Instant;
use std::env;
use smoltcp::Error;
use smoltcp::wire::{EthernetAddress, IpAddress};
use smoltcp::iface::{ArpCache, SliceArpCache, EthernetInterface};
use smoltcp::socket::{SocketSet};

fn main() {
    utils::setup_logging();

    let opts = getopts::Options::new();
    let matches = opts.parse(env::args().skip(1)).unwrap();
    let device = RawSocket::new(&matches.free[0]).unwrap();
    let startup_time = Instant::now();

    let arp_cache = SliceArpCache::new(vec![Default::default(); 8]);
    let mut iface = EthernetInterface::new(
        Box::new(device), Box::new(arp_cache) as Box<ArpCache>,
        EthernetAddress([0x00, 0x00, 0x00, 0x00, 0x00, 0x01]),
        [IpAddress::v4(10, 0, 0, 1)]);

    let mut sockets = SocketSet::new(vec![]);

    loop {
        let timestamp = Instant::now().duration_since(startup_time);
        let timestamp_ms = (timestamp.as_secs() * 1000) + (timestamp.subsec_nanos() / 1000000) as u64;
        match iface.poll(&mut sockets, timestamp_ms) {
            Ok(()) | Err(Error::Exhausted) => (),
            Err(e) => debug!("poll error: {}", e)
        }
    }
}
```

Now we can create a veth pair:

```
ip link add veth0 type veth peer name veth1
ip link set dev veth0 up
ip link set dev veth1 up
```

We run the client:
```
target/debug/examples/arp veth0
```

Let's send some ping with a wrong destination mac:

```
# we add a dummy ip in the 10.0.0.0/24 subnet on veth1 to force linux to send ping through it
ip addr add 10.0.0.10/24 dev veth1
# we add a static arp entry to force echo request to be sent with a wrong mac address
ip neigh change 10.0.0.1 dev veth1 lladdr aa:bb:cc:dd:ee:ff nud permanent
```

Now we can ping:

```
~ ❯❯❯ ping 10.0.0.1
PING 10.0.0.1 (10.0.0.1) 56(84) bytes of data.
64 bytes from 10.0.0.1: icmp_seq=1 ttl=64 time=0.282 ms
64 bytes from 10.0.0.1: icmp_seq=2 ttl=64 time=0.349 ms
^C
--- 10.0.0.1 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1023ms
rtt min/avg/max/mdev = 0.282/0.315/0.349/0.037 ms
```
Here is the packet capture, just change the extension to `.pcap` to open it.
[out.txt](https://github.com/m-labs/smoltcp/files/820108/out.txt)

I'm not sure this is *really* an issue, and I could not find an RFC that explicitely states that a host should reject ethernet frames with a wrong destination address. Still, it seems to me that the correct behavior would be to ignore such packets.

whitequark, 2017-03-06:
This is a real issue, sure.

whitequark, 2017-03-06:
Fixed in master. Multicast packets are always ignored though.

closed 2017-03-06 by whitequark