smoltcp-rs/smoltcp #16: Errors parsing ICMP Destination Unreachable

adamgreig, 2017-06-15:
When smoltcp receives an ICMPv4 Destination Unreachable Port Unreachable, it tries to parse the payload as an IPv4 packet, but this often fails with Truncated because the original packet was much longer than whatever initial portion of it was included with the ICMP response. This error (from https://github.com/m-labs/smoltcp/blob/master/src/wire/icmpv4.rs#L384) then gets bubbled up to the top `poll()` call where it looks like a network error.

As far as I can tell it shouldn't actually be an error though, since the RFC says you just include the first 64 bytes (in practice apparently Linux does a few hundred?) of the original packet, but you don't expect to see all of it. smoltcp parsing an IPv4 packet errors out if the packet buffer isn't long enough to contain the length the IPv4 header promises, so the whole thing stops. Not sure what the neatest solution is, perhaps having a way to parse the IPv4 header without caring that the payload likely won't be long enough?

whitequark, 2017-06-15:
> Not sure what the neatest solution is, perhaps having a way to parse the IPv4 header without caring that the payload likely won't be long enough?

Yeah. I actually thought that's we were doing already, but guess not. You can use Ipv4Packet instead of Ipv4Repr here to avoid excessive length checks (we might need to add a new constructor though).

adamgreig, 2017-06-16:
`icmpv4::Repr` already constructs an `Ipv4Packet` first, then constructs the `Ipv4Repr` from it, and the `Ipv4Repr` is only used by returning it as part of the `icmpv4::Repr::DstUnreachable`.

I guess the question is what `Repr::DstUnreachable` should contain -- right now it gets `Ipv4Repr` to indicate the original header, plus eight bytes of the original ipv4 payload. Having it contain an `Ipv4Packet` instead seems troublesome since `Ipv4Packet` contains a reference to the original buffer, rather than just a copy of the pertinent header information.

I don't _think_ being able to construct an `Ipv4Repr` where the payload buffer is shorter than the header-indicated length is immediately an issue since `Ipv4Repr` doesn't thereafter store any reference to the payload. In other words, I'm not sure what benefit the payload length check in `Ipv4Repr::parse()` gives, except to maybe catch some errors early (but make it impossible to construct a header from a partial packet, as is needed here). Of course the `Ipv4Packet` given to `parse()` must be long enough to contain a full IPv4 header, but I think needn't be any longer.

whitequark, 2017-06-17:
> In other words, I'm not sure what benefit the payload length check in Ipv4Repr::parse() gives

It's to prevent panics when using accessors in Ipv4Packet. smoltcp is supposed to be panic-free.

whitequark, 2017-06-17:
Anyway, can you give me a pcap file with a packet that fails to parse?

adamgreig, 2017-06-17:
Sure, I've sent one your way.

whitequark, 2017-06-17:
Attaching it to the issue: [icmpunreachable.zip](https://github.com/m-labs/smoltcp/files/1082564/icmpunreachable.zip)

closed 2017-06-21 by whitequark

birkenfeld, 2018-06-20:
This seems to have resurfaced, since `Ipv4Packet::new_checked()` is now used there, which *does* check the total_len field.

Reopen this or create a new issue?

dlrobertson, 2018-06-20:
I *think* ensuring the packet is larger than 8 + the end of the destination address should be good enough for now. I don't see anything that should panic in [icmpv4.rs:415-434](https://github.com/m-labs/smoltcp/blob/master/src/wire/icmpv4.rs#L415-L434) or [icmpv6.rs:483-498](https://github.com/m-labs/smoltcp/blob/master/src/wire/icmpv6.rs#L483-L498).

We also need to make sure we don't cause problems when we're parsing the IP payload in a destination unreachable message for ICMP sockets. The +8 should take care of that unless the next header isn't the UDP payload. I'm not entirely sure what to do in that case. I'll look into adding a test for that, so any updates to fix this issue don't cause problems for ICMP sockets.

commit 818fa7dd (2017-06-21): Unbreak parsing of ICMP unreachable messages.

Fixes #16.