smoltcp-rs/smoltcp #17: UDP sending reads uninitialised memory to check buffer length with unpredictable results

adamgreig, 2017-06-15:
I've been chasing down intermittent panics "undersized payload: Truncated" while sending UDP packets and I think I've gotten to the bottom of it.

The gist is that in `socket/udp.rs:264` when we try to make a new `UdpPacket` from payload, the payload is a TX buffer obtained from `Device`, with the Ethernet and IPv4 headers set up, but without any UDP header set yet. `UdpPacket::new` then makes a `UdpPacket` from this buffer and checks the buffer length is at least `packet.len()`, a method which reads the UDP header length field. However, this hasn't been set yet!

So, at startup this is often 0 and works fine. During most of my operations, this is set to the previous packet length (and they're all the same) and works fine. Once I have to respond to an ARP request, this is most recently set to 0xC0A8, the first two octets of my desktop's IP address. Since the buffer being transmitted is not that long, we get a panic.

Slightly longer explanation of where this comes from:
1. User calls `Interface::poll()`
2. `poll` calls `Interface::emit()` to send new packets
3. `Interface::emit()` calls `Socket::dispatch` for each socket
4. In `socket/udp.rs:238 Socket::dispatch()` we dequeue a PBuf and put it into a `UdpRepr` called `repr`
5. `Socket::dispatch()` calls the `emit` closure from `Interface::emit()` with `payload` set to `UdpRepr` as an `IpPayload`
6. Back in `interface/ethernet.rs:383 Interface::emit()`, we obtain a TX buffer from Device and turn it into a `Frame`
7. The frame has its ethernet and IPv4 headers set and an `Ipv4Packet` made
8. `payload: IpPayload` (our `UdpRepr` from before with the PBuf to send) has its `emit()` called with the `Ipv4Packet` payload buffer (inside the TX buffer from Device), called `payload`
9. Now `socket/udp.rs:264 impl IpPayload for UdpRepr emit()` makes a new `UdpPacket` with `buffer` set to `payload`, which is the Ipv4Packet from 8 and has not had any UDP fields initialised
10. Now `wire/udp.rs:37 new()` makes a `Packet` (`UdpPacket`) from `buffer`
11. And finally calls `packet.len()`, which reads the LENGTH field from `buffer`

I think `UdpPacket::new()` shouldn't check `packet.len()` in this case, but I'm not sure how else it can find out what length UDP packet is meant to go into it. The `IpPayload emit()` in `socket/udp.rs`, on success, then calls `UdpPacket::emit` which will do a `copy_from_slice` to copy the PBuf into the TX buffer.

I also haven't checked to see if the same problem exists for TCP but I'm yet to hit any panics from it...

whitequark, 2017-06-15:
It's kind of amazing that I managed to put in an uninitialized read bug without ever using `unsafe`...

adamgreig, 2017-06-15:
I guess "uninitialized" is a little sloppy: it was initialised once, to something, but not in the context of this use, and not as a length. Not technically memory unsafe... ;)

whitequark, 2017-06-15:
It's not memory unsafe but it's functionally identical to uninitialized memory read bugs in a language where such reads are defined behavior.

adamgreig, 2017-06-16:
Yes, that's true. It certainly has the same feel as classic uninitialised reads in C.

On further inspection this actually happens before the UDP packet construction, in the IPv4 packet construction. After making the `EthernetFrame` and setting its src, dst, and ethertype fields, we make a new `Ipv4Packet` with that `Frame`'s buffer, and `Ipv4Packet::new` has a length check against `packet.header_len()` which.. reads from the uninitialised buffer. So this _does_ affect all IPv4 packets being sent.

Happily the `EthernetFrame::new` only checks the length is at least long enough to contain the whole Ethernet frame header (a static check against `field::PAYLOAD.start`), and does not read the buffer, so is OK.

I wonder if the various `Packet::new` constructors should _not_ check against packet lengths, and instead those checks should be moved into the `emit()` methods, which have access to both the uninitialised buffer _and_ the new data to be copied in to it, which will have the correct length fields.

adamgreig, 2017-06-16:
This also means all `Packet`s constructed using `new()` with an uninitialised buffer (i.e. all outgoing packets?) have a bunch of methods that all read fields from the buffer and then return junk information, until the buffer is filled in by some `Repr`'s `emit()` method being given the `Packet.payload()` to fill.

whitequark, 2017-06-21:
> This also means all Packets constructed using new() with an uninitialised buffer (i.e. all outgoing packets?) have a bunch of methods that all read fields from the buffer and then return junk information

This is actually fine and intended; all these methods are specifically written so that they will never panic. This matters, of course, because you can receive any sort of random junk from the network.

I'm still chewing through the rest... I am impressed with your debugging skills.

whitequark, 2017-06-21:
> On further inspection this actually happens before the UDP packet construction, in the IPv4 packet construction. After making the EthernetFrame and setting its src, dst, and ethertype fields, we [...]

Hmm, I don't quite see it. Let's look at this piece of code:
```
let mut frame = try!(EthernetFrame::new(&mut tx_buffer));
frame.set_src_addr(src_hardware_addr);
frame.set_dst_addr(dst_hardware_addr);
frame.set_ethertype(EthernetProtocol::Ipv4);

repr.emit(frame.payload_mut());

let mut ip_packet = try!(Ipv4Packet::new(frame.payload_mut()));
payload.emit(&repr, ip_packet.payload_mut());
```

After the `repr.emit()` call, the frame payload has the entire IP header, including the payload length, filled in. So I don't see why the `Ipv4Packet::new` call would ever fail (we might want to use `expect` there instead).

adamgreig, 2017-06-22:
[`repr.emit(frame.payload_mut())`](https://github.com/m-labs/smoltcp/blob/master/src/iface/ethernet.rs#L396) calls [`IpRepr::emit`](https://github.com/m-labs/smoltcp/blob/master/src/wire/ip.rs#L284) which calls [`Ipv4Packet::new`](https://github.com/m-labs/smoltcp/blob/master/src/wire/ipv4.rs#L108) which calls [`Ipv4Packet::header_len`](https://github.com/m-labs/smoltcp/blob/master/src/wire/ipv4.rs#L132) which reads from the buffer, if I'm following the chain of events correctly.

whitequark, 2017-06-22:
Oh *that* is how it's happening. Hrm. Yes I see now.

adamgreig, 2017-06-22:
The second `Ipv4Packet::new` in `EthernetDevice`'s closure should never fail (since we've literally just done the same thing inside `IpRepr::emit` and with an `expect` there(!)) so that one might be good for `expect` instead.

whitequark, 2017-06-22:
Yes, I was about to make this change (we have something like 60 instances of `(Packet|Frame)::new` and many of those never fail) but ultimately I think what we should do is rather split `new` into two functions...
  * `(Packet|Frame)::parse` which does perform the same amount of checking as we do now. This would be used for packets from the network.
  * `(Packet|Frame)::wrap`, which does not. It may `debug_assert!` on cases that don't require interrogation of the buffer but the utility seems pretty low. This would be used for packets we're constructing ourselves, and for packets that we know don't need the length checks, such as ones we've just constructed recently.

This is basically what you were suggesting all the time, only now I fully understand the reasons behind.

Naming bikeshed welcome, I don't particularly like `parse` nor `wrap`.

adamgreig, 2017-06-22:
Maybe just `new` and `new_checked`? The "parsing" as such is just to check the length of the buffer is OK. Or `wrap` and `wrap_checked`.

I had a quick look over the rest of the code, I don't think `Frame::new` will ever cause problems. I think the following `Packet::new` uses suffer from this issue:

* [`wire/ip.rs:284`](https://github.com/m-labs/smoltcp/blob/818fa7dda409e8ff35be48d5a00bbd833f2163e9/src/wire/ip.rs#L284)
* [`wire/icmpv4.rs:446`](https://github.com/m-labs/smoltcp/blob/818fa7dda409e8ff35be48d5a00bbd833f2163e9/src/wire/icmpv4.rs#L446)
* [`socket/tcp.rs:1134`](https://github.com/m-labs/smoltcp/blob/818fa7dda409e8ff35be48d5a00bbd833f2163e9/src/socket/tcp.rs#L1134)
* [`socket/udp.rs:205`](https://github.com/m-labs/smoltcp/blob/818fa7dda409e8ff35be48d5a00bbd833f2163e9/src/socket/udp.rs#L205)
* [`iface/ethernet.rs:398`](https://github.com/m-labs/smoltcp/blob/818fa7dda409e8ff35be48d5a00bbd833f2163e9/src/iface/ethernet.rs#L398)
* [`iface/ethernet.rs:425`](https://github.com/m-labs/smoltcp/blob/818fa7dda409e8ff35be48d5a00bbd833f2163e9/src/iface/ethernet.rs#L425)

Might not be an exhaustive list but I think it can only apply to uses in the transmit chain which filters it down pretty quickly.

whitequark, 2017-06-22:
> I don't think Frame::new will ever cause problems

The reason I want to have *every* `new` function split is that in the transmit chain you never actually want to propagate the error upwards. If you get an `Err` from `new` then it's a logic error, and smoltcp should panic.

adamgreig, 2017-06-22:
On second thoughts `new` and `new_unchecked` might actually be more ergonomic, given most uses are of the checked variety.

> you never actually want to propagate the error upwards. If you get an Err from new then it's a logic error, and smoltcp should panic.
:+1:

whitequark, 2017-06-22:
> On second thoughts new and new_unchecked might actually be more ergonomic, given most uses are of the checked variety.

I think it's important to *not have a default case* here. You have to actively consider whether the buffer contents and size can be trusted or not.

commit 6f6494a0 (2017-06-24): Do not attempt to validate length of packets being emitted.

This is a form of an uninitialized read bug; although safe it caused
panics. In short, transmit buffers received from the network stack
should be considered uninitialized (in practice they will often
contain previously transmitted packets or parts thereof). Wrapping
them with the only method we had (e.g. Ipv4Packet) treated the buffer
as if it contained a valid incoming packet, which can easily fail
with Error::Truncated.

This commit splits every `fn new(buffer: T) -> Result<Self, Error>`
method on a `Packet` into three smaller ones [... cut]
commit de809b04 (2017-06-24): Do not attempt to validate length of packets being emitted.

This is a form of an uninitialized read bug; although safe it caused
panics. In short, transmit buffers received from the network stack
should be considered uninitialized (in practice they will often
contain previously transmitted packets or parts thereof). Wrapping
them with the only method we had (e.g. Ipv4Packet) treated the buffer
as if it contained a valid incoming packet, which can easily fail
with Error::Truncated.

This commit splits every `fn new(buffer: T) -> Result<Self, Error>`
method on a `Packet` into three smaller ones [... cut]
commit 735def7e (2017-06-24): Do not attempt to validate length of packets being emitted.

This is a form of an uninitialized read bug; although safe it caused
panics. In short, transmit buffers received from the network stack
should be considered uninitialized (in practice they will often
contain previously transmitted packets or parts thereof). Wrapping
them with the only method we had (e.g. Ipv4Packet) treated the buffer
as if it contained a valid incoming packet, which can easily fail
with Error::Truncated.

This commit splits every `fn new(buffer: T) -> Result<Self, Error>`
method on a `Packet` into three smaller ones [... cut]
closed 2017-06-24 by whitequark