smoltcp-rs/smoltcp #24: Use timestamp for TCP initial sequence number

whitequark, 2017-06-26:
We don't have a clock with the 4 us resolution as RFC 793 wants, but it's likely fine to use the 1 ms resolution clock too.

klickverbot, 2017-06-26:
Also see RFC 1948/6528 – timestamps have been augmented by a PRNG since to avoid sequence number attacks.

whitequark, 2017-06-26:
@klickverbot Is there some source of truth for which RFCs are actually authoritative for TCP? RFC 793 is hopelessly outdated and has errata, RFC 1122 fixes some of that, highlights a few common errors, many of which I did make, but also piles completely useless junk on top of it (I think every ICMP message it specifically mentions except unreachables and echo request/reply is deprecated, strongly discouraged and ignored by any modern implementation).

klickverbot, 2017-06-26:
@whitequark: Unfortunately, I don't know of any up to date list of RFCs relevant for the various areas, but I found the review in RFC 7414 to be quite useful (from 2015).

whitequark, 2017-06-26:
@klickverbot Thanks

astro, 2018-03-05:
The timeouts for sending out IGMP reports could also use a PRNG.

Could one be built with just an `Instant` timestamp as input?

whitequark, 2018-03-05:
The TCP specification dictates a particular algorithm for the PRNG. While not great it's more complex than just the timestamp...