smoltcp-rs/smoltcp #36: Hardware Checksum Calculation

steffengy, 2017-08-30:
I think the possibility to move checksum calculation to the hardware when possible,
would be neat.

E.g. With an STM32x and LWIP you can do
and use `Checksum insertion control` to let the hardware handle those checksums.

```c
#define CHECKSUM_BY_HARDWARE
#define CHECKSUM_GEN_IP 0
#define CHECKSUM_GEN_UDP 0
#define CHECKSUM_GEN_TCP 0
#define CHECKSUM_GEN_ICMP 0
#define CHECKSUM_CHECK_IP 0
#define CHECKSUM_CHECK_UDP 0
#define CHECKSUM_CHECK_TCP 0
#define CHECKSUM_CHECK_ICMP 0
```

whitequark, 2017-08-30:
Sure. Do you have any particular hardware in mind? Show me the datasheet and I'll think about exposing this capability.

steffengy, 2017-08-30:
STM32F2xxx/F4xxx (F4 is just a superset with more functionality basically, ethernet is afaik the same):

http://www.st.com/content/ccc/resource/technical/document/reference_manual/3d/6d/5a/66/b4/99/40/d4/DM00031020.pdf/files/DM00031020.pdf/jcr:content/translations/en.DM00031020.pdf

Searching for "checksum" does already give good results.

steffengy, 2017-08-31:
To clarify, since it wasn't really clearly worded:

The only thing that smoltcp would have to support for this, 
would be disabling checksum generation (use zeros as checksum).

The hardware will generate & insert the checksum on its own before sending the frame.

As shown above LWIP (an ip stack) supports fine-grained control of checksums e.g. disabling them for ICMP, ...

Not sure how to best expose this in rust, using features would lead to a "feature overload".

whitequark, 2017-08-31:
Yeah, I understood. I haven't had time to flesh out the ideas I had, hence the silence.

whitequark, 2017-09-08:
OK, so the route to fixing this would be to:
  * introduce a `enum Checksum { Both, Rx, Tx, None }`;
  * add a parameter `verify_cksum: Checksum` to all `*Repr::{parse,emit}` methods;
  * rename `DeviceLimits` into `DeviceCaps` or something (for "capabilities");
  * add fields `soft_ethernet_cksum: Checksum, soft_ip_cksum: Checksum, ..` to `DeviceCaps`;
  * thread all this through `EthernetInterface`.
I don't have time to do this right now but would gladly accept a PR.

closed 2017-10-02 by whitequark