smoltcp-rs/smoltcp #51: Implementing IGMP protocol and multicast support

podhrmic, 2017-10-06:
Hello, for our application I need smoltcp to handle multicast packets (the application communicates  over multicast). I am happy to do that and implement this functionality to smoltcp, but I would like to get some pointers and guidance first.

1. IGMP protocol will have to be implemented - I would add another branch on the match expression here https://github.com/m-labs/smoltcp/blob/master/src/iface/ethernet.rs#L249 
2. Implement `EthernetProtocol::IGMP` with underlying logic
3. when IGMP (and multicast) is enabled we check `dst_addr()` against the list of multicast MAC addresses we are subscribed to (here: https://github.com/m-labs/smoltcp/blob/master/src/iface/ethernet.rs#L244 )
4. appropriately handle timeouts (see https://en.wikipedia.org/wiki/Internet_Group_Management_Protocol ) - not sure if we need a separate timer or we should use the time supplied to `iface.poll()`

 Does it sound like a good approach to you?

whitequark, 2017-10-06:
Sounds good! I suggest working on it in small pieces and seeking review early, as I try to ensure the quality of code in smoltcp stays very high.

whitequark, 2018-09-22:
@podhrmic This is done, right?

podhrmic, 2018-09-24:
Yes, I believe IGMP support was added in #178

closed 2018-09-24 by whitequark