smoltcp-rs/smoltcp #54: Would a dedicated crate for packet decoding make sense?

little-dude, 2017-10-11:
Hi,

The `wire` module provides abstractions for packet decoding, which could also be useful outside of `smoltcp`. For example, I am considering implementing the OpenFlow protocol and I'd like to re-use what you already did for packet parsing and add support OpenFlow headers.

In case I'm unclear, I'd like to see something similar to Golang's [gopacket/layers](https://godoc.org/github.com/google/gopacket/layers) library.

whitequark, 2017-10-11:
> The wire module provides abstractions for packet decoding, which could also be useful outside of smoltcp.

Why not just depend on smoltcp without any of its features?

little-dude, 2017-10-11:
I can totally do that. But don't you mind the wire module having code for protocols that are not really used in `smoltcp` ? For instance, would you accept a PR that add support for decoding OpenFlow headers?

kpcyrd, 2017-10-11:
somewhat offtopic, but [pktparse-rs](https://github.com/bestouff/pktparse-rs) might be an alternative.

whitequark, 2017-10-15:
@little-dude Sure why not, if it's up to smoltcp's code quality. We can always split it afterwards.

@kpcyrd That doesn't provide constructors.

whitequark, 2017-10-25:
@little-dude OK, looking closer at [gopacket/layers] I now see what you mean. It is indeed massive.

Here's what I suggest:

  * You depend on smoltcp for (e.g.) its `ip::checksum` function, `IpAddress` enum, and so on.
  * You decode OpenFlow in your own `wire` module in your own crate. The parsing and serialization itself in smoltcp are extremely lightweight--they do not use some combinators or macros or so on, so there isn't really any code to depend on.
  * Once there's a nontrivial demand of a library like [gopacket/layers] we branch that out into its own crate, or otherwise decide.

Right now I don't really have the bandwidth to review everything, but I don't see why we can't do this a bit later when we have more contributors.

[gopacket/layers]: https://godoc.org/github.com/google/gopacket/layers

little-dude, 2017-10-25:
Thanks @whitequark.
Openflow itself being massive, I started a new crate with smoltcp as dependency, like you said.

> Once there's a nontrivial demand of a library like gopacket/layers we branch that out into its own crate, or otherwise decide.

That would be nice yeah :) `gopacket` is a great library. As far as I know [libpnet](https://docs.rs/pnet/) is the closest thing that exists in Rust so far.

@kpcyrd that's a cool library for decoding packets, but as @whitequark said, I also need constructors to send packets.

closed 2017-10-25 by little-dude

whitequark, 2017-10-25:
I've considered building smoltcp on top of libpnet but I really dislike their heavy-handed type-based approach. As you can see smoltcp keeps things very simple.