smoltcp-rs/smoltcp #56: rustfmt.toml for smoltcp

podhrmic, 2017-10-13:
Would be possible to add a `rustfmt.toml` file (see https://github.com/rust-lang-nursery/rustfmt/blob/master/Configurations.md ) specifying the preferred formatting for smoltcp? 

That way it would be easy to keep consistent formatting.

For example I noticed that the default rustfmt settings are different than what smoltcp is using, so it is hard to automatically format existing files.

whitequark, 2017-10-13:
Sure. Do you think you could play with rustfmt's options to get an output very close to the current one? Then I'll merge that and reformat the rest.

podhrmic, 2017-10-13:
I ll give it a try. I think the main difference is the alignment - i.e. this is what is in smoltcp now:

```Rust
    /// Return the header length.
    /// The result depends on the value of the message type field.
    pub fn header_len(&self) -> usize {
        match self.msg_type() {
            Message::EchoRequest    => field::ECHO_SEQNO.end,
            Message::EchoReply      => field::ECHO_SEQNO.end,
            Message::DstUnreachable => field::UNUSED.end,
            _ => field::CHECKSUM.end // make a conservative assumption
        }
    }
```

And this is after `rustfmt` with default settings:

```Rust
    /// Return the header length.
    /// The result depends on the value of the message type field.
    pub fn header_len(&self) -> usize {
        match self.msg_type() {
            Message::EchoRequest => field::ECHO_SEQNO.end,
            Message::EchoReply => field::ECHO_SEQNO.end,
            Message::DstUnreachable => field::UNUSED.end,
            _ => field::CHECKSUM.end, // make a conservative assumption
        }
    }
```

So I am not sure how much you care about the right alignment?

whitequark, 2017-10-13:
To be honest I find that it helps readability a lot.

podhrmic, 2017-10-13:
After poking around it looks like there is no setting for rustfmt that would keep the LHS => RHS alignment as you have it. Apparently it is a discouraged style for Rust. 

So the question is - do you prefer to keep your current formatting even though it means no automatic format checking, or do you want to switch to the format mentioned above?

whitequark, 2017-10-13:
Let's keep it, but just so that your work doesn't go nowhere, please post your rustfmt config here. If we ever switch to using rustfmt I'll make use of it.

podhrmic, 2017-10-13:
OK. Place `rustfmt.toml` in the root dir of the project.

The contents are simple and can be further tuned:
```
attributes_on_same_line_as_field = false
attributes_on_same_line_as_variant = false
struct_field_align_threshold = 20
tab_spaces = 4
```

See https://github.com/rust-lang-nursery/rustfmt for details

whitequark, 2017-10-15:
I looked at the diff produced by rustfmt. It is 11937 line long and readability suffers, particularly badly with tests. Thank you for your work but I don't think I'm going to use rustfmt for my projects.

closed 2017-10-15 by whitequark