smoltcp-rs/smoltcp #62: TCP hangs when sending

pothos, 2017-10-25:
Hello,
actually I wanted to debug it, but need some help because it is disappears e.g. if writing to pcap or extensive logging is used.

The smoltcp server thread on tap0 will send in 64 byte steps, a Linux socket client thread will read only to ~292800 bytes instead of reading all 10000000 bytes. Smoltcp is handing out keep-alives from this point on. The minimal example is here:

https://github.com/pothos/taptest

`cargo run --release` should reproduce the behavior.

`RUST_LOG=trace cargo run --release` should let it _sometimes_ complete (same as adding print statements in the sending server).
Uncommenting the pcap writer will let it complete reliably.

I have experienced it with other values as well, but hope that this combination also reproduces it on other machines.

Best regards,
Kai

whitequark, 2017-10-25:
Oh yeah you hit a bug; retransmit timer gets reset by ACKs whereas it should not.

pothos, 2017-10-26:
Thanks for looking at it, now I have to read the RFCs along with the code. Because I don't understand yet why only `ack_len` is used to dequeue and a change in the window size is not relevant(?). Rightnow I don't see where the retransmit would be involved… I'll continue tomorrow.

pothos, 2017-10-28:
Ok, finally got it working. Besides keeping the retransmission timer there was an overflow that needed to be fixed. Preparing a PR now.

whitequark, 2017-12-21:
@pothos I'm finally making progress on this—see d1e2292 for one method I'm using to expose underlying bugs. I am afraid that your fix is hiding one of them.

whitequark, 2017-12-22:
@pothos I've integrated a test quite like your `taptest` in the commit 44db954. I've managed to expose one other bug as well.

whitequark, 2017-12-22:
Fixed in b247f64.

closed 2017-12-22 by whitequark

pothos, 2018-01-02:
Thanks for including the fix - I am happy that the wrapping can be covered without distinction of cases ;)