smoltcp-rs/smoltcp #68: ARP storm

jordens, 2017-11-01:
smoltcp seems to be generating a bit of an ARP storm, at least when interacting with seom security scan appliance.

dlrobertson, 2017-11-01:
Also related to #50 and #25

klickverbot, 2017-11-01:
We saw a similar issue in our network a couple of weeks ago (ARP requests for the same address more than once every millisecond, from a nominally idle KC705), but it was late at night trying to get some actual data, so we just restarted the core device and it was gone. Unfortunately, I can't seem to find the dump right now, but reproducing this might definitely be a bit non-straightforward, as the problem didn't occur again. No security scan appliances on the subnet that I'm aware of.

whitequark, 2017-11-01:
This is the exact same issue as #25.

@klickverbot It is not easily reproducible in practice but it's trivial to write tests for, so it's not a problem. Don't bother with the dumps, I am well aware of all causes of this one.

dhslichter, 2017-11-14:
@whitequark what is the timeline for fixing this issue?  This is a high priority for us because we can't use ARTIQ 3 on our network until it is addressed (the ARP storm breaks some other groups' stuff), and I think other labs may be in a similar position.

whitequark, 2017-11-15:
@dhslichter I plan to have it fixed in 2 days.

sbourdeauducq, 2017-11-19:
@whitequark ping

whitequark, 2017-11-22:
Done

closed 2017-11-22 by whitequark