smoltcp-rs/smoltcp #71: [REPORT] not working on macOS

LuoZijun, 2017-11-07:
Line: https://github.com/m-labs/smoltcp/blob/master/src/phy/sys/raw_socket.rs#L21

hi, the `libc::AF_PACKET` constant only work on linux kernel.

whitequark, 2017-11-07:
I know. Feel free to contribute a macOS port of the hosted components. I do not currently have plans for that.

Of course, you can always disable the raw and TAP sockets using `default-features = false` in your `Cargo.toml`.

LuoZijun, 2017-11-08:
ok, thx for your reply.

LuoZijun, 2017-12-26:
@whitequark hi, i almost finish `rust-bpf` (macOS), do u interesting if i make one PR ?

https://github.com/LuoZijun/exodus/blob/master/netif/src/bpf.rs

whitequark, 2017-12-26:
Sure, if you clean up that code it seems like a perfect fit for our `phy` module.

LuoZijun, 2018-01-31:
See PR [136](https://github.com/m-labs/smoltcp/pull/136) .

PR #268 (2018-12-28): implement BPF sockets to be used as RawSocketDesc on BSD-like OSes
I added a file under `phy/sys` named `bpf.rs` and put all the code there so you can review the code easily. I also made other small changes so as to be able to build and run on macOS and FreeBSD with the feature `phy-raw_sockets`. I also named the methods of the `BpfDevice` struct same as in the `RawSocketDesc` struct, and reimport `BpfDevice` as `RawSocketDesc`.

Another point that can be discussed about my implementation is about the `interface_mtu` method (see the comment in the code).

I tested the code with the `tcpdump` e [... cut]
closed 2018-12-28 by whitequark