smoltcp-rs/smoltcp #74: Bug in tcp option parsing?

phil-opp, 2017-11-08:
I think there is a bug in [this line](https://github.com/m-labs/smoltcp/blob/4ae84ab36a32f8e0c530d968bac9207cbf0513b3/src/wire/tcp.rs#L535) in wire/tcp.rs:

```rust
length = *buffer.get(1).ok_or(Error::Truncated)? as usize;
let data = buffer.get(2..length).ok_or(Error::Truncated)?;
```

I think it should be `buffer.get(2..(2+length))` instead, because otherwise we try to take the range `2..1` for an option with length 1.

whitequark, 2017-11-08:
Nope: https://play.rust-lang.org/?gist=bb7d8c35c1d97d35c88a8919498866e3&version=stable

See the `fuzz/` directory in the source? That takes care of this silly class of bugs. In fact I think that's why I use `get` there.

closed 2017-11-08 by whitequark

phil-opp, 2017-11-15:
I didn't mean that `get(2..1)` is a bug itself. In the DHCP protocol an option with length 4 has 6 octets (4 + code + length fields) and I thought that TCP was identical. But it seems like in TCP the length field includes the whole option, i.e. length 4 means 4 octets. So no bug here, just a strange difference between the TCP and DHCP option format.