smoltcp-rs/smoltcp #77: [ASK] Ethernet Frame struct

LuoZijun, 2017-11-09:
Code Lines: https://github.com/m-labs/smoltcp/blob/master/src/wire/ethernet.rs#L88-L91

The Wiki: https://en.wikipedia.org/wiki/Ethernet_frame#Structure

as you can see, wikipedia said ethernet frame after `payload` filed, have one `Frame check sequence (32‑bit CRC)` , but i have not see in we `ethernet frame struct` .

whitequark, 2017-11-09:
The frame check sequence, just like the preamble and start delimiter, are handled by hardware.

whitequark, 2017-11-09:
Note the difference between ethernet _packet_ and ethernet _frame_; the packet includes preamble and FCS, whereas the frame does not.

LuoZijun, 2017-11-09:
@whitequark 

[image]

thx, the wikipedia lie to me -_-

LuoZijun, 2017-11-09:
@whitequark 

now the wikipedia page [Ethernet_frame](https://en.wikipedia.org/wiki/Ethernet_frame) have been fixed 😄

closed 2017-11-09 by LuoZijun

whitequark, 2017-11-09:
Looks like the Wikipedia people disagree: https://en.wikipedia.org/w/index.php?title=Ethernet_frame&diff=809483347&oldid=809464039
Oh well. In any case, it's virtually always handled by hardware, so there is no point in exposing it to software.

LuoZijun, 2017-11-10:
okay