smoltcp-rs/smoltcp #82: Add support for tun interface?

raintean, 2017-11-16:


whitequark, 2017-11-16:
What's the point?

LuoZijun, 2017-11-19:
@whitequark i think his mean is platform api ( `phy` ) .

whitequark, 2017-11-20:
Yes. Tap interfaces are software emulation of layer 2 interfaces (Ethernet in our case) so we can handle them with EthernetInterface. However, tun interfaces are software emulation of layer 3 interfaces (TCP/UDP/IP directly), so we would need an alternative to EthernetInterface that works on layer 3. We can implement that, but why should we?

LuoZijun, 2017-11-20:
@whitequark as i know, the `phy` module is `RAW_SOCKET` api work on  bsd/linux/windows platforms. 

why we need `phy` module ? beacuse we need send some ethernet frame packet(tap) or ip packet(tun)  to `netif` device (just like `iface` module).

in unix-like system, they have 3 types `netif` device: loopback, tap and tun.

for now, we just impl `loopback` and `tap` on linux platform, if we has implemented `tun` device, that is good for some VPN user.

of course, they can use third party lib, like (meh/rust-tun)[https://github.com/meh/rust-tun] .

whitequark, 2017-11-20:
If you just want the `phy::tun` module, I am fine with that. Please feel free to implement it and submit as a PR, and I will merge it.

LuoZijun, 2017-11-20:
@whitequark okay, i try.

liranringel, 2017-12-23:
@LuoZijun any progress?
Let me know if you're still into it.

LuoZijun, 2017-12-25:
@liranringel 

https://github.com/LuoZijun/exodus/blob/master/netif/src/bpf.rs

i'm try port `bpf` on macOS.

this also can also resolve [issues/71](https://github.com/m-labs/smoltcp/issues/71)

LuoZijun, 2018-01-31:
@liranringel 

See [PR-136](https://github.com/m-labs/smoltcp/pull/136) .

Dirbaio, 2022-05-18:
TUN is supported since 0.8

closed 2022-05-18 by Dirbaio