smoltcp-rs/smoltcp #87: Closure based polling on ethernet interface

phil-opp, 2017-11-30:
At the moment, polling the ethernet interface works by passing in a `SocketSet`, which is passed down to the `process_*` methods (e.g. `process_udp`). The `process_*` methods then iterate over the socket set and pass the packet to all sockets that accept it.

The problem with this approach is that the socket interface requires buffering in the sockets and polling the sockets later, which is undesirable in some situations. For example, if we send a control command to a microcontroller, we want it to react as soon as it receives the packet and avoid any latency caused by buffering and polling.

Therefore I would propose to change the signature of the `EthernetInterface::poll` method, so that it takes a closure:

```rust
fn poll<F>(&mut self, f: F, timestamp: u64) -> Result<Option<u64>>
    where F: FnOnce(IpAddress, Port, Packet)
```

(The exact types for `Port` and  `Packet` still need to be determined.)

This closure can then be passed down to the `process_*` methods and invoked for each packet. This would allow the user to react immediately to certain packets and e.g. put the other ones in socket buffers. We could also add a `poll_sockets` method with the current behavior (which calls `poll` with a closure that iterates over a socket set).

What do you think about such a change?

cc @oli-obk

whitequark, 2017-12-02:
I don't think FnOnce works in this signature.

phil-opp, 2017-12-04:
> I don't think FnOnce works in this signature.

Why not? The function would only handle a single packet per call (like it does today), so it would only invoke the closure once.

> It's really unclear to me how this is supposed to work lifetime-wise too...

I try to implement a prototype. Where do you expect lifetime related problems?

whitequark, 2017-12-04:
> Why not? The function would only handle a single packet per call (like it does today), so it would only invoke the closure once.

How so? Right now poll() will pull all packets that are available from the device and process them.

phil-opp, 2017-12-04:
Ok, then I misunderstood the interface and we need `FnMut`.