smoltcp-rs/smoltcp #93: Steady ARP breeze

jonaskeller, 2017-12-12:
The flood prevention for ARP requests (#25) does its job. Our FPGA running Artiq 3.1 now only sends out ARP requests at 1Hz instead of >1kHz.

However, since it never gets an answer, it seems to keep going indefinitely (I reset it after >70k attempts). I doubt that this is going to be an issue for anyone on our network, but stopping after a certain amount of consecutive, unanswered requests would probably be a more elegant solution (if it can be implemented with reasonable resources).

whitequark, 2017-12-12:
> stopping after a certain amount of consecutive, unanswered requests would probably be a more elegant solution (if it can be implemented with reasonable resources).

It's probably trying to close the TCP connection. Can you set the log level to TRACE and show me the core device log for this issue?

jonaskeller, 2017-12-12:
Sure, I'll just have to wait for it to occur again. Based on what I've seen so far, that might take a few days.

jonaskeller, 2017-12-15:
Just an update: Last time it got triggered, it stopped after ~10 requests (without ever getting a reply). I'll keep monitoring it, but given the low rate of trigger events, it might take a long time before I get a chance to observe it again. If you have any hints on how to provoke it, I'm happy to try things out.

whitequark, 2017-12-15:
Oh, I think I know what happens there, retransmits in TIME-WAIT state. I'll try and see if I can reproduce this soon enough.

whitequark, 2018-02-16:
@jonaskeller Any luck reproducing?

jonaskeller, 2018-02-16:
Looking at ARP packets from the FPGA in all packet captures since then, I don't see an instance besides those mentioned above. I have to admit that I haven't been constantly monitoring it, though - this is more a side-effect of looking into other issues.

whitequark, 2018-02-16:
OK. I don't see any potential cause in the codebase anymore. Reopen if this reappears.

closed 2018-02-16 by whitequark