smoltcp-rs/smoltcp #95: Create Repr and/or Packet Builder types

dlrobertson, 2017-12-17:
### `Repr` builders

As more options begin to surface for each packet type, creating a packet `Repr` becomes more cumbersome to use. In addition, when a new field is added to a `Repr` a large portion of the codebase needs to be changed (see #59). It may be worthwhile to create `Repr` builder types. This also allows us to enforce sensible defaults (e.g. the user of the builder doesn't have to know the default hop limit should be `0x40`).

The implementation of this builder should be a non-consuming builder such that the following is possible.

```rust
let mut builder = ReprBuilder::new();
builder.something(value);
let repr = builder.finalize();
builder.something(other_value);
let other_repr = builder.finalize();
```

### `Packet` builders

The typical implementation of `Repr` blurs the lines between a packet builder and a high level representation of a packet, due to the presence of `emit`. When the end goal is to use `Repr::emit` we could short circuit the `finalize` step if the builder also has an `emit` method.

Current code may look like the following
```rust
let packet = Packet::new_checked(&mut bytes)?;
let repr = Repr { something: value };
repr.emit(packet);
```

If builders are created this could look like the following
```rust
let packet = Packet::new_checked(&mut bytes)?;
let repr = ReprBuilder::new().something(value).finalize();
repr.emit(packet)
```

If the builder has `emit`
```rust
let packet = Packet::new_checked(&mut bytes)?;
ReprBuilder::new().something(value).emit(packet);
```

### Open questions

Still not entirely sure how I feel about this since the `Repr` is a packet builder. Would it be possible/beneficial to break out the Packet builder from the `Repr`? Such that when we recv `Packet -> Repr` and when we xmit `PacketBuilder -> Packet`.

whitequark, 2017-12-17:
How would `Repr` builders even work for enum `Repr`s

dlrobertson, 2017-12-17:
Great question. Honestly, I didn't consider the enum `Repr` case. I guess you could have something like `ReprBuilder::new(<enum type>)` that returns a different builder for each enum type? Otherwise I guess you'd have to store a `type` member and all of the optional data required for each variant. Either way isn't ideal.

whitequark, 2017-12-17:
IMO there is no particular need for this, and in light of the `enum` case there seems no obvious elegant solution either. I am open to reconsideration if a future use case comes up, of course.

closed 2017-12-17 by whitequark