smoltcp-rs/smoltcp #96: FPGA closes TCP connection on RPC, terminating the experiment

jonaskeller, 2017-12-19:
Running Artiq 3.1, nist_qc2, Windows 7 64bit

We have an experiment which runs in a loop, updating some datasets and running scheduler.check_pause() on each iteration. Every few minutes, it terminates with the error message

    Terminating with exception (ConnectionResetError: [WinError 10054] An existing connection was forcibly closed by the remote host)

It seems to be caused by a timeout when the ARP entry for the host is invalidated and a new request needs to be sent out on the first of the RPCs. See [core log](https://github.com/m-labs/smoltcp/files/1572443/TCP_issue2_RST_from_FPGA_corelog.txt), around 439s. Stripping the code to a minimal example for this isn't as straightforward as I hoped, however. Let me know if you need more information or testing.

whitequark, 2017-12-19:
Please file ARTIQ issues against the ARTIQ repository.

closed 2017-12-19 by whitequark

jonaskeller, 2017-12-19:
Sorry - will do that next time.