smoltcp-rs/smoltcp #97: segment not in receive window in server example

crawford, 2017-12-21:
I was playing around with the server example and noticed some strange failures. When using netcat, I was able to successfully receive the "hello" message when connecting to port 6969. When using curl, however, I start seeing "segment not in receive window" repeatedly.

To set up my experiment, I cloned the repo at master (960b0012) and ran the following on my machine (taken from the docs):

```
sudo ip tuntap add name tap0 mode tap user $USER
sudo ip link set tap0 up
sudo ip addr add 192.168.69.100/24 dev tap0
```

After that, I start the server with `cargo run --example server tap0` and start my client with `curl 192.168.69.1:6969`.

Curl exists immediately:

```
$ curl 192.168.69.1:6969
hello
```

But the server gets stuck:

```
$ cargo run --example server tap0
...
[    10.102s] (server): tcp:6969 send greeting                                              
[    10.102s] (server): tcp:6969 close        
[    10.102s] (socket::tcp): #1:192.168.69.1:6969:192.168.69.100:50030: state=ESTABLISHED=>FIN-WAIT-1
[    10.102s] (socket::tcp): #1:192.168.69.1:6969:192.168.69.100:50030: outgoing segment will send data or flags
[    10.102s] (socket::tcp): #1:192.168.69.1:6969:192.168.69.100:50030: tx buffer: sending 6 octets at offset 0
[    10.102s] (socket::tcp): #1:192.168.69.1:6969:192.168.69.100:50030: sending FIN|ACK
[    10.102s] -> EthernetII src=02-00-00-00-00-01 dst=xx-xx-xx-xx-xx-xx type=IPv4
                \ IPv4 src=192.168.69.1 dst=192.168.69.100 proto=TCP
                 \ TCP src=6969 dst=50030 fin seq=1227067479 ack=3067899883 win=0 len=6
[    10.143s] <- EthernetII src=xx-xx-xx-xx-xx-xx dst=02-00-00-00-00-01 type=IPv4
                \ IPv4 src=192.168.69.100 dst=192.168.69.1 proto=TCP
                 \ TCP src=50030 dst=6969 seq=3067899883 ack=1227067486 win=29200 len=0
[    10.143s] (socket::tcp): #1:192.168.69.1:6969:192.168.69.100:50030: received ACK of FIN
[    10.143s] (socket::tcp): #1:192.168.69.1:6969:192.168.69.100:50030: state=FIN-WAIT-1=>FIN-WAIT-2
[    10.143s] (socket::tcp): #1:192.168.69.1:6969:192.168.69.100:50030: tx buffer: dequeueing 6 octets (now 0)
[    10.308s] <- EthernetII src=xx-xx-xx-xx-xx-xx dst=02-00-00-00-00-01 type=IPv4
                \ IPv4 src=192.168.69.100 dst=192.168.69.1 proto=TCP
                 \ TCP src=50030 dst=6969 seq=3067899882 ack=1227067486 win=29200 len=0
[    10.308s] (socket::tcp): #1:192.168.69.1:6969:192.168.69.100:50030: segment not in receive window (3067899882..3067899882 not intersecting 3067899883..3067899883), will send challenge ACK
[    10.308s] -> EthernetII src=02-00-00-00-00-01 dst=xx-xx-xx-xx-xx-xx type=IPv4
                \ IPv4 src=192.168.69.1 dst=192.168.69.100 proto=TCP
                 \ TCP src=6969 dst=50030 seq=1227067486 ack=3067899883 win=0 len=0
[    10.716s] <- EthernetII src=xx-xx-xx-xx-xx-xx dst=02-00-00-00-00-01 type=IPv4
                \ IPv4 src=192.168.69.100 dst=192.168.69.1 proto=TCP
                 \ TCP src=50030 dst=6969 seq=3067899882 ack=1227067486 win=29200 len=0
[    10.716s] (socket::tcp): #1:192.168.69.1:6969:192.168.69.100:50030: segment not in receive window (3067899882..3067899882 not intersecting 3067899883..3067899883), will send challenge ACK
[    10.716s] -> EthernetII src=02-00-00-00-00-01 dst=xx-xx-xx-xx-xx-xx type=IPv4
                \ IPv4 src=192.168.69.1 dst=192.168.69.100 proto=TCP
                 \ TCP src=6969 dst=50030 seq=1227067486 ack=3067899883 win=0 len=0
...
```

I suspect this is due to the fact that curl immediately fills the receive window, but I don't know enough about TCP to know where the off-by-one error arises.

whitequark, 2017-12-21:
Thanks for the report.

whitequark, 2017-12-21:
This is not actually a bug, but a side effect of the fact that `server.rs` never reads from `tcp:6969`. As a result, the TCP in your kernel is ever trying (unsuccessfully) to push the rest of the data sent by curl in. Since `server.rs` also never aborts the connection (it only closes its half-connection) this can go on indefinitely. There is no off-by-one involved.

Wireshark can help you understand this:
[image]

closed 2017-12-21 by whitequark

whitequark, 2017-12-21:
I've improved the log message in 3347bbe.

crawford, 2017-12-25:
Ah, that makes sense. Thank you!