smoltcp-rs/smoltcp #103: Processing of IPv6 packets

dlrobertson, 2017-12-24:
Metabug detailing roadmap to add processing of IPv6 packets
 
- [ ] Additions to `wire`
   - [ ] [ICMPv6]
   - [ ] [MLDv2]
   - [ ] [NDISC]
   - [ ] Handle extension headers
       - [ ] Add [IPv6] `Options
       - [ ] Add [IPv6] `Hop-by-Hop` extension header
       - [ ] Add [IPv6] `Routing` extension header
       - [ ] Add [IPv6] `Fragment` extension header
       - [ ] Add [IPv6] `No Next` extension header
 - [ ] Updates to `socket`
   - [ ] Update `IcmpSocket` to also handle [ICMPv6] packets
   - [ ] Update `RawSocket` to also handle [IPv6] packets
 - [ ] Updates to `iface`
   - [ ] Correctly handle [IPv6] packets, forwarding packets to the `accept`ing socket.
   - [ ] Register to multicast addresses on initialization (See [ADDRARCH 2.7.1] and [NDISC])
       - [ ] All-nodes
       - [ ] Solicited-node address for each IPv6 address assigned to the interface
   - [ ] Add Address Resolution via [NDISC]
   - [ ] Add Neighbor Unreachability Detection via [NDISC]

[ICMPv4]: https://tools.ietf.org/html/rfc792
[ICMPv6]: https://tools.ietf.org/html/rfc4443
[IPv6]: https://tools.ietf.org/html/rfc8200
[MLDv2]: https://tools.ietf.org/html/rfc3810
[NDISC]: https://tools.ietf.org/html/rfc4861\
[ADDRARCH 2.7.1]: https://tools.ietf.org/html/rfc4291#section-2.7.1

dlrobertson, 2017-12-24:
The very first PR should probably just handle IPv6 packets and forward them to the socket. The second should probably handle sending packets via `RawSocket`. After those are done then we can begin to tacke address resolution etc.
 
For a first pass at extension headers it may be worthwhile to add the extension headers to `wire` but only handle the `Pad1` and `PadN` options. Discarding packets with options according to the options type [RFC 8200 4.2]

[RFC 8200 4.2]: https://tools.ietf.org/html/rfc8200#section-4.2

hjr3, 2018-01-15:
@dlrobertson let me know what help you need, if any, on this.

dlrobertson, 2018-01-15:
The two next big steps will be to add the following two packet types to `wire`.

 - ICMPv6
 - Extension headers (start with just the `Pad1` and `PadN` `Options`)

Both of which are a bit of a quest. I've just started on ICMPv6 support, so I'd suggest working on extension headers (However, if you're an ICMPv6 wizard or dead set on working on it I'm okay with pivoting to Extension Headers). #99 is also another great issue that is related to this, but it might be better to do #99 post ICMPv6 support.

If you need some pointers, feel free to ping me as `dlrobertson` on the `freenode` `#m-labs` channel or the [Redox OS chat]. I'm definitely a bit more active on the [Redox OS chat] `#networking` channel, but @whitequark is also available on the `freenode` `#m-labs` IRC channel.

[Redox OS chat]: https://github.com/redox-os/book/blob/master/src/contributing/chat.md

hjr3, 2018-01-15:
@dlrobertson I will start on the extension headers.

dlrobertson, 2018-01-15:
Awesome! I definitely appreciate the help

hjr3, 2018-01-31:
@dlrobertson I have not been able to catch you on #m-labs irc and I just requested access to redox chat. I have started working on the `Hop-by-Hop Options Header` implementation at https://tools.ietf.org/html/rfc8200#section-4.3.

dlrobertson, 2018-07-11:
Still a lot more to be done, but #252 was the last issue that was part of this.

closed 2018-07-11 by dlrobertson