smoltcp-rs/smoltcp #104: Implement TCP Fast Retransmit

whitequark, 2017-12-25:
Right now we'll be blasting the entirety of our buffer before even considering retransmitting, which is silly.

barskern, 2018-04-23:
I would like to take a stab at implementing this feature. I have been programming in Rust for some months now, and I've taken a subject in University about networks and protocols. I've started to read through the code, but I was wondering if I could get a helping indication on where to start.

whitequark, 2018-04-23:
There's a comment that says `// FIXME: implement fast retransmit`, that would be a good start :)

barskern, 2018-04-24:
I've been reading the documentation, and started to dig into the code. Today I've been trying to write a test to isolate a situation where a fast retransmit would occur, however it doesnt seem to be right. Currently the test looks like this:

```rust
   #[test]
    fn test_fast_retransmit_after_triple_duplicate_ack() {
        let mut s = socket_established();
        s.remote_win_len = 6;
        
        s.send_slice(b"abcdef123456ABCDEFZYXWVU").unwrap();

        // This packet is supposed to not be acknowledged (recived)
        recv!(s, time 1000, Ok(TcpRepr {
            seq_number: LOCAL_SEQ + 1,
            ack_number: Some(REMOTE_SEQ + 1),
            payload:    &b"abcdef"[..],
            ..RECV_TEMPL
        }));
        recv!(s, time 1005, Ok(TcpRepr {
            seq_number: LOCAL_SEQ + 1 + 6,
            ack_number: Some(REMOTE_SEQ + 1),
            payload:    &b"123456"[..],
            ..RECV_TEMPL
        }));
        recv!(s, time 1010, Ok(TcpRepr {
            seq_number: LOCAL_SEQ + 1 + 6 + 6,
            ack_number: Some(REMOTE_SEQ + 1),
            payload:    &b"ABCDEF"[..],
            ..RECV_TEMPL
        }));
        recv!(s, time 1015, Ok(TcpRepr {
            seq_number: LOCAL_SEQ + 1 + 6 + 6 + 6,
            ack_number: Some(REMOTE_SEQ + 1),
            payload:    &b"ZYXWVU"[..],
            ..RECV_TEMPL
        }));
        // First ACK
        send!(s, time 1020, TcpRepr {
          seq_number: REMOTE_SEQ + 1,
          ack_number: Some(LOCAL_SEQ + 1),
          window_len: 6,
          ..SEND_TEMPL
        });
        // Second (duplicate) ACK
        send!(s, time 1025, TcpRepr {
          seq_number: REMOTE_SEQ + 1,
          ack_number: Some(LOCAL_SEQ + 1),
          window_len: 6,
          ..SEND_TEMPL
        });
        // Third (duplicate) ACK
        // Should trigger a fast retransmit of dropped packet
        send!(s, time 1030, TcpRepr {
          seq_number: REMOTE_SEQ + 1,
          ack_number: Some(LOCAL_SEQ + 1),
          window_len: 6,
          ..SEND_TEMPL
        });
        // Fast retransmit packet
        recv!(s, time 1035, Ok(TcpRepr {
            seq_number: LOCAL_SEQ + 1,
            ack_number: Some(REMOTE_SEQ + 1),
            payload:    &b"abcdef"[..],
            ..RECV_TEMPL
        }));
    }
```

However at https://github.com/m-labs/smoltcp/blob/master/src/socket/tcp.rs#L870-L888 the `ack_number` and `self.local_seq_no` are still equal. I would expect `ack_number` to be less than `self.local_seq_no` because the reciver has not yet recived the first packet.

I am trying to create a test-case which emulates the following diagram:

[image]

Can someone with a better understanding of the library help me develop this testcase?

whitequark, 2018-05-14:
Sorry for dropping the ball on this, and I'm glad you have been able to resolve the issue!

> I haven't refined my ideas yet, but I was thinking about proposing a separate state struct which would control both flow control and congestion control.

Sounds good.

barskern, 2018-05-14:
I have now implemented fast retransmit :tada: 

It is done by adding two new fields to the `TcpSocket` which track the last recived ACK number and the amount of duplicated ACKs recived. When the third duplicated ACK is detected, it uses `self.timer.set_for_retransmit(timestamp)` with the current `timestamp` to schedule a faster retransmit. This requires few lines of code, however it is a bit ambigious _imo_.

As previously mentionned, I was thinking about creating a new struct which could encapsulate these checks and provide a clean interaction between the actual socket and the flow control and congestion control.

I was thinking of making a bigger issue detaling the thoughts and ideas I currently have and ask for some opinions. Should I make a pull requst on the currently working implementiation or should this issue be closed in favor of the bigger implementation of flow control and congestion control? Making a pull request now would give the library a temporary implementation of TCP fast retransmit while the other is being developed.

barskern, 2018-05-15:
Okay a PR is coming!

One small thing though is that when using `self.set_for_retransmit(timestamp)`, it adds a retransmit delay. Hence the retransmit is unnessecarily waiting for that delay.

I could add a new state to the `Timer` which would explicitly handle retransmitting without the delay.

What are your thoughts @whitequark?

```
#0:fe80::1:80:fe80::2:49500: duplicate ACK number 2 for 10007
#0:fe80::1:80:fe80::2:49500: do a fast retransmit for 10007
#0:fe80::1:80:fe80::2:49500: retransmitting at t+0.105sms
```

whitequark, 2018-05-15:
A `Timer::FastRetransmit` state seems good!

commit 375b0266 (2018-05-17): Implement fast retransmit for TcpSocket

Implement a simple way to discover and respond to three duplicate ACKs
from the reciver of the data. This leads to a fast retransmit of the
last unacknowledged seqment. This is feature is described in [RFC 5681].

Closes: #104

Closes: #214
Approved by: whitequark
closed 2018-05-17 by m-labs-homu