smoltcp-rs/smoltcp #109: About example tcpdump

LuoZijun, 2017-12-28:
[image]

why has `checksum incorrect` error ?

this packet send by system, not me.

whitequark, 2017-12-28:
TCP checksum offload.

LuoZijun, 2017-12-28:
@whitequark 

https://github.com/m-labs/smoltcp/blob/master/src/wire/ip.rs#L777

but why checksum is invalid ?

i think this shoud't happend.

maybe recv raw packet `( socket(AF_PACKET, ...) ) ` has some mistake ?

whitequark, 2017-12-28:
Google "TCP checksum offload". Basically, the OS puts zero in the checksum field, and since tcpdump is receiving packets from the OS, it also gets packets with such timestamps. But the NIC, before transmitting the packet, calculates the actual timestamp and substitutes it just in time.

LuoZijun, 2017-12-28:
okay, thank u let me know that.

vbox doing that, i guess.

whitequark, 2017-12-28:
Yes, hypervisors are typically capable of handling checksum offload.

closed 2017-12-28 by whitequark