smoltcp-rs/smoltcp #111: FIN within the window closes the connection but the sequence number indicates unseen data

pothos, 2018-01-03:
Observed in some cases using RAW sockets with the HTTP example where the data packet and the FIN packet arrive in a different order (or the data packet could also be lost). Then nothing is received because the socket is already in CLOSE-WAIT.

Here are logs of fail and success cases, first the trace messages and then `tcpdump -r` for the pcap log:
https://gist.github.com/pothos/a788ba6c58d916bb12d071b881f7ab03

pothos, 2018-01-03:
A fix would be just to ignore it and send an ACK but maybe it can also be remembered since the assembler is there already.

Is this test a good starting point?
```
    #[test]
    fn test_established_not_fin() {
        let mut s = socket_established();
        send!(s, TcpRepr {
            control: TcpControl::Fin,
            seq_number: REMOTE_SEQ + 10,
            ack_number: Some(LOCAL_SEQ + 1),
            ..SEND_TEMPL
        }, Ok(Some(TcpRepr {
            seq_number: LOCAL_SEQ + 1,
            ack_number: Some(REMOTE_SEQ + 1),
            ..RECV_TEMPL
        })));
        assert_eq!(s.state, State::Established);
    }
```

whitequark, 2018-01-03:
The assembler doesn't know about flags. Your test looks good.

pothos, 2018-01-03:
Yes, just meant that it would be consistent with the assembler to remember it for a faster acknowledgement in the end. But for the beginning it makes sense to handle it the same way as those not in the receive window i.e. `segment_in_window = false`?

whitequark, 2018-01-03:
Yeah.

Not having reassembly means a severe perf degradation; not handling these out-of-order FINs doesn't mean much because it just means sockets spend a little longer closing, so I think this should not be a priority.

pothos, 2018-01-03:
I saw you assinged your self as well, but in case you did not start yet you can have a look on the PR ;) Thanks

commit fcffa6a5 (2018-01-05): Disregard TCP FIN flag if it arrives in a segment not at window start.

Fixes #111.
closed 2018-01-05 by whitequark