smoltcp-rs/smoltcp #115: Implementing RxToken and TxToken with lifetimes

crawford, 2018-01-11:
I'm trying to implement the `Device` trait in some firmware that I'm writing but I'm stumped. I'm trying to create the RxToken and TxToken such that they reference entries in an underlying DMA pool and therefore need explicit lifetimes. Here is a sample that exhibits the same compilation errors:

```rust
use smoltcp::{self, phy};

pub struct RxToken<'a> {
    buffer: &'a [u8],
}

impl<'a> phy::RxToken for RxToken<'a> {
    fn consume<R, F>(self, _timestamp: u64, f: F) -> smoltcp::Result<R>
    where
        F: FnOnce(&[u8]) -> smoltcp::Result<R>,
    {
        panic!("Not implemented")
    }
}

pub struct TxToken<'a> {
    buffer: &'a [u8],
}

impl<'a> phy::TxToken for TxToken<'a> {
    fn consume<R, F>(self, _timestamp: u64, _len: usize, _f: F) -> smoltcp::Result<R>
    where
        F: FnOnce(&mut [u8]) -> smoltcp::Result<R>,
    {
        panic!("Not implemented")
    }
}

pub struct MAC { }

impl<'a> phy::Device<'a> for MAC {
    type RxToken = RxToken<'a>;
    type TxToken = TxToken<'a>;

    fn capabilities(&self) -> phy::DeviceCapabilities {
        phy::DeviceCapabilities::default()
    }

    fn receive(&'a mut self) -> Option<(Self::RxToken, Self::TxToken)> {
        None
    }

    fn transmit(&'a mut self) -> Option<Self::TxToken> {
        None
    }
}
```

I get the following compilation error when I attempt to pass a `MAC` into `EthernetInterfaceBuilder::new()`:

```
error[E0277]: the trait bound `for<'d> &MAC: smoltcp::phy::Device<'d>` is not satisfied
  --> src/main.rs:40:21
   |
40 |     let mut iface = EthernetInterfaceBuilder::new(&device)
   |                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `for<'d> smoltcp::phy::Device<'d>` is not implemented for `&MAC`
   |
   = help: the following implementations were found:
             <MAC as smoltcp::phy::Device<'a>>
   = note: required by `<smoltcp::iface::InterfaceBuilder<'b, 'c, DeviceT>>::new`
```

I haven't be able to implement `Device` when the associated types also have lifetimes. I feel like I'm missing something obvious with the `impl` block. Any help you can offer would be greatly appreciated.

phil-opp, 2018-01-11:
I think `EthernetInterfaceBuilder` requires ownership of the device, so try to pass `device`instead of a shared reference `&device`.

crawford, 2018-01-11:
Oh my, that was the issue. 😱 Thank you so much for the pointer.

closed 2018-01-11 by crawford