smoltcp-rs/smoltcp #120: Route broadcast packets directly

kaendfinger, 2018-01-13:
When sending a packet to 255.255.255.255, routing should send directly, instead of to the gateway.
This is essentially a PR, except it's so trivial, I decided just to do an issue instead.

[Problem Code](https://github.com/m-labs/smoltcp/blob/master/src/iface/ethernet.rs#L875)

## Current

```rust
    fn route(&self, addr: &IpAddress) -> Result<IpAddress> {
        // Send directly.
        if self.in_same_network(addr) {
            return Ok(addr.clone())
        }
```

## Fixed

```rust
    fn route(&self, addr: &IpAddress) -> Result<IpAddress> {
        // Send directly.
        if self.in_same_network(addr) || addr.is_broadcast() {
            return Ok(addr.clone())
        }
```

whitequark, 2018-01-15:
FYI it's quite a bit less work to merge this as a PR.

closed 2018-01-15 by whitequark

kaendfinger, 2018-01-16:
Sorry about that @whitequark, I was on a slow network at the time, and cloning was taking a long time.

commit cb5be488 (2018-01-15): Route broadcast packets directly.

Closes #120.