smoltcp-rs/smoltcp #127: smoltcp doesn't replace unspecified src addresses with local addresses for UDP and TCP sockets

batonius, 2018-01-18:
**Problem:** Right now binding a UDP or a TCP socket to `0.0.0.0:*` results in packets being sent with src IP `0.0.0.0`, not with the IP assigned to the interface, even though ICMP and Raw sockets do the replacement.

**Cause:** Both `UdpSocket::dispatch` and `TcpSocket::dispatch` use `IpRepr::Unspecified` and set `src_addr` to `local_endpoint`, wich is either `IpAddress::Ipv4` or `IpAddress::Ipv6`, even if the IP addr in question is itself unspecified, i.e. `IpAddress::Ipv4(0, 0, 0, 0)`. `IpRepr::lower` treat such combinations as not requiering the replacement: https://github.com/m-labs/smoltcp/blob/master/src/wire/ip.rs#L498 .

**Possible solutions:** 
1. Change `IpRepr::lower` to consider case `IpRepr::Unspecified(src_addr: IpAddr::Ipv*(src_addr), ..) if src_addr.is_unspecified()` as an equivalent to `IpRepr::Unspecified(src_addr: IpAddr::Unspecified, ..)`. I've done it in https://github.com/batonius/smoltcp/commit/8e4422cee017f60232cab7078efa683a70a733c9 , but it breaks `TcpSocket` since it uses `IpRepr::lower` with empty address list: https://github.com/m-labs/smoltcp/blob/master/src/socket/tcp.rs#L1315 . I'm not sure how to proceed from here - we can either provide some IPs to `IpRepr::lower` in `TcpSocket` or add a special case for empty `fallback_src_addrs` in `IpRepr::lower`.
2. Change `UdpSocket::dispatch` and `TcpSocket::dispatch` to check `src_addr` and replace it with `IpAddr::Unspecified`. This would fix the problem, but I'm not sure if there are other places where `IpRepr::Unspecified` could be used with `IpAddr::Ipv4`/`IpAddr::Ipv6` with an unspecified addr.

**Comment:** I was gonna fix it myself, but I'm not sure which of the options would be the optimal, so I'm creating the issue first.

batonius, 2018-01-26:
@whitequark Any ideas?

whitequark, 2018-01-26:
How about: solution 1, but you just don't replace `0.0.0.0` / `::` with a specified IP address if there are no IP addresses provided?

batonius, 2018-01-26:
Sure, I'll make a PR.

PR #133 (2018-01-26): Replace unspecified src IP addrs with the interface IP
As discussed in #127, `IpRepr::lower` should treat `IpRepr::Unspecified(src_addr: IpAddr::Ipv*(src_addr), ..) if src_addr.is_unspecified()` as `IpRepr::Unspecified(src_addr: IpAddr::Unspecified, ..)`, i.e. try to replace `src_addr` with the concrete IP address assigned to the interface, unless `fallback_src_addrs` is empty, in which case it should keep the unspecified `src_addr`.

@dlrobertson I replaced the code for IPv6 case with an equivalent of the IPv4 one for uniformity and because the fix is more clear this way.

Closes #127
commit 283109ca (2018-01-26): Treat unspecified IPv4/IPv6 addresses as IpAddress::Unspecified in IpRepr::lower.

Closes #127.
closed 2018-01-26 by whitequark