smoltcp-rs/smoltcp #129: Routing, network address translation

canndrew, 2018-01-23:
Is this out of scope for this library? I'm implementing some of this stuff myself in a library that I plan to port to smoltcp. I'm wondering if I should try to move my implementations upstream at some point.

whitequark, 2018-01-23:
Unclear. I'd need to look at your proposal to see if it makes sense in smoltcp.

canndrew, 2018-01-23:
For example, there could be a `Nat<D>` type which wraps a `Device` and implements `Device`. Any IP packets sent from the underlying device get their source endpoint mapped to some global endpoint before being sent from the `Nat`. And vice-versa for incoming packets sent to the `Nat`.

A `Router` type could be constructed from a set of `Device`s and a set of routes. As you drive the router, any frame that gets read from one device gets dropped or written to another.

batonius, 2018-01-23:
I would like to see routing in `smoltcp`, but I think it should be implemented on top of (or as a part of) a more general support for multiple interfaces, something we lack right now, otherwise, it makes little sense. There were some (quite vague) ideas discussed in #55, which overlap with your proposal to a significant extent, but I haven't come up with a specific design yet, so maybe you could contribute to that discussion as well.

whitequark, 2018-01-23:
I agree with @batonius.

astro, 2018-03-25:
> For example, there could be a Nat<D> type which wraps a Device and implements Device. Any IP packets sent from the underlying device get their source endpoint mapped to some global endpoint before being sent from the Nat. And vice-versa for incoming packets sent to the Nat.

In Linux NAT happens in the outgoing device path (`POSTROUTING` chain). There the outgoing address can be used naturally.

Dirbaio, 2020-12-25:
Hello! As part of issue cleanup I'm closing inactive or already-answered "question" issues, since they're not actionable.

Note this doesn't mean questions aren't welcome. They still are! It's just that they shouldn't stay open forever.

closed 2020-12-25 by Dirbaio