smoltcp-rs/smoltcp #137: poll_at returns the time left to call poll instead of the timestamp at which to call poll

dlrobertson, 2018-01-31:
In working on #124 I've noticed that the socket implementations of `poll_at` often return 0 instead of a timestamp at which to call poll. This can happen for example if the `rx_buffer` for a `UdpSocket` is not empty (See [src/socket/udp.rs:254](https://github.com/m-labs/smoltcp/blob/master/src/socket/udp.rs#L254)). [`EthernetInterface::poll_at`](https://github.com/m-labs/smoltcp/blob/master/src/iface/ethernet.rs#L326-L330) will then return `Some(0)` due to this value being the minimum value, which is obviously a duration, but the docs state that the value should be an instant.

I can roll this up into my work for #124, if this is indeed an issue.

whitequark, 2018-01-31:
It's not obviously a duration. It means "as early as possible", and is used in situations where the current timestamp is inaccessible.

dlrobertson, 2018-01-31:
Ah, that makes sense. I guess I was thinking that if that was the case the timestamp would be equal to the timestamp passed as an argument. Thanks for the clarification

closed 2018-02-07 by dlrobertson