smoltcp-rs/smoltcp #144: Add function `netmask()` and `broadcast` on Ipv4Cidr/Ipv6Cidr ?

LuoZijun, 2018-02-03:
Add function `netmask()` and `broadcast()` on Ipv4Cidr/Ipv6Cidr ?

```rust

impl Ipv4Cidr {
    pub fn netmask(&self) -> Address {
        let mut addr = [0u8; 4];
        if self.prefix_len >= 8 {
            for idx in 0..((self.prefix_len - self.prefix_len % 8) / 8) {
                addr[idx] = 255;
            }
        }

        Address(addr)
    }

    pub fn broadcast(&self) -> Address {
        let mut addr = [0u8; 4];
        let mask = self.netmask;
        for idx in 0..4 {
            if mask[idx] == 255 {
                addr[idx] = self.address.0[0];
            } else {
                addr[idx] = 255;
            }
        }
        Address(addr)
    }
}
```

whitequark, 2018-02-03:
Only on IPv4Cidr, IPv6 does not have netmasks and it only has a single broadcast address, `::`.

LuoZijun, 2018-02-04:
See (PR-146)[https://github.com/m-labs/smoltcp/pull/146]

closed 2018-02-07 by LuoZijun