smoltcp-rs/smoltcp #147: Use standard library types where possible

canndrew, 2018-02-05:
I've noticed smoltcp defines its own `Ipv4Address` and `IpEndpoint` types instead of using the standard `Ipv4Addr` and `SocketAddr` types. Is there any good reason for this? It's kind of annoying to have to convert between the two.

whitequark, 2018-02-05:
`#[no_std]`

canndrew, 2018-02-05:
Oh, right.

Couldn't we define the types to be identical to the standard library types and then only use them if `#[no_std]` is being used, otherwise just re-export the standard library types?

whitequark, 2018-02-05:
The standard library types have a different API that doesn't follow the way every other type in smoltcp is designed. Read the documentation for the wire module to see what I mean.

whitequark, 2018-02-05:
That said, I would accept a PR that adds `From` and `Into` implementations, where possible.

canndrew, 2018-02-05:
Here you go: https://github.com/m-labs/smoltcp/pull/149

PR #149 (2018-02-05): Add conversions to/from std wire types
Fixes #147
closed 2018-02-05 by whitequark