smoltcp-rs/smoltcp #156: Add simpler and more robust APIs for creating packets

canndrew, 2018-02-08:
I recently ported a project over to using the wire types from smoltcp. The places in my code where I create Arp/Udp/Ipv4/Ethernet packets are now significantly more verbose and I also introduced several bugs that could have been prevented by a more typesafe API.

Basically the problem is this: Suppose I want to wrap a udp packet into an ipv4 packet. The way I'd currently do this is with something like.

```rust
let udp: UdpPacket<_> = ...;

let udp_raw_bytes = udp.into_inner();
let ipv4_repr = Ipv4Repr {
    src_addr: ...,
    dst_addr: ...,
    protocol: IpProtocol::Tcp,
    payload_len: udp_raw_bytes.len(),
};
let mut ipv4 = Ipv4Packet::new(allocate_some_bytes(ipv4_repr.buffer_len()));
ipv4_repr.emit(&mut ipv4);
ipv4.payload_mut().clone_from_slice(&udp_raw_bytes);
```

This is too long. Also it contains a couple of bugs that aren't caught by the typechecker or prevented by the API. Firstly, by converting the udp packet into raw bytes we lose information (at the type level) about what sort of data we're putting in the packet. It would be better if there was a packet-emitting function that took a `UdpPacket` directly so that I can't accidentally wrap one as a TCP packet. Also, I have to tell the API the lengths of data I'm dealing with in several places, and it's possible to get these wrong (by eg. calling `ipv4_repr.buffer_len()` thinking it will give me the total length of memory I need to allocate for the packet). The library already knows how much memory it needs, I shouldn't have to tell it. Also it's easy to forget to copy across the UDP packet data. This also shouldn't be possible to do.

I propose to add packet-creation APIs that look something like this:

```rust
Ipv4Packet::from_udp<T, B>(
    src_addr: Ipv4Address,
    dst_addr: Ipv4Address,
    ttl: u8,
    packet: UdpPacket<T>,
    alloc: impl FnOnce(usize) -> B,
) -> Ipv4Packet
where
    T: AsRef<[u8]>,
    B: AsMut<[u8]>,
```

That way, I can pass it the allocation function I want to use and trust it to create the packet correctly. Would something like this be desirable?

whitequark, 2018-02-08:
The API you propose is a non-starter because it includes an alloc function. smoltcp never, ever, allocates anything. This is literally in the second paragraph of README.

whitequark, 2018-02-08:
> This API would allow me to write Ipv4Packet::from_udp(..., |n| my_embedded_system_grab_n_bytes_from_ring_buffer(n)).

Have you actually tried to write this? The ownership doesn't work out. Consider what will happen if you will try to use that hypothetical `my_embedded_system_grab_n_bytes_from_ring_buffer` function to allocate both the backing storage for UdpPacket and for Ipv4Packet.

whitequark, 2018-02-08:
(Also, why "my_embedded_system"? Embedded systems have heaps, and it makes perfect sense to not use an allocator in a high-throughput user-mode stack.)

canndrew, 2018-02-08:
> Have you actually tried to write this? The ownership doesn't work out. Consider what will happen if you will try to use that hypothetical `my_embedded_system_grab_n_bytes_from_ring_buffer` function to allocate both the backing storage for UdpPacket and for Ipv4Packet.

I'm not sure what you mean. You will allocate twice - which is bad - but sometimes necessary. I was thinking about raising another issue to allow packets to backed by non-contiguous storage, but in my case just cloning some bytes here and there will be good enough.

> Also, why "my_embedded_system"?

It was just an example of where you might not want to use a malloc function because you might not have access to one or it might be too slow.

canndrew, 2018-02-08:
Maybe the proposed API isn't ideal but do you at least see the problems that I'm gesturing at in the bug report? How the current API is cumbersome and bug-prone for some use-cases?

whitequark, 2018-02-08:
> You will allocate twice - which is bad - but sometimes necessary.

You can't do this with safe code only because the alloc function returns a mutable pointer into the ring buffer. I'm not interested in adding APIs that require the use of unsafe code elsewhere.

> I was thinking about raising another issue to allow packets to backed by non-contiguous storage

Packets already can be backed by non-contiguous storage though. The only difference is you have to use `new` instead of `new_checked`, and in case of `Repr`, use an empty slice as payload.

> How the current API is cumbersome and bug-prone for some use-cases?

It's not "cumbersome and bug-prone", it's "too general". What you're introducing is *a* specialized wrapper for a fairly narrow use case, and I'm not convinced that particular wrapper is good for inclusion for the reasons I've outlined above.

LuoZijun, 2018-02-08:
> There's no way to create a EthernetFrame<B> where B stores the data in several distinct chunks. This is what I'm referring to.

Why not store data in a buffer ？

whitequark, 2018-02-08:
> I think we're talking past each other. I just want to return a zero-initialized BytesMut. You don't need unsafe code for that.

You can't return several mutable pointers into a ring buffer without unsafe code. That your suggested API can work with Vec::new is not enough.

> There's no way to create a EthernetFrame<B> where B stores the data in several distinct chunks. This is what I'm referring to.

Why would you want that? If you want to use scatter-gather capabilities, just use separate buffers for EthernetFrame, Ipv4Packet, UdpPacket, etc. You can do that right now. Splitting the 14-byte Ethernet header over multiple buffers does not appear useful to me.

> Yes, being able to do things that are plain wrong - like wrapping a UDP packet as a TCP packet - is too general.

Actually, *being able* to do this is an important feature of the `Packet` layer; there is intentionally no checks other than (optional) length checks and no complicated magic to make sure you don't make some mistake. In case of the `Repr` layer, you can't do anything that's plain wrong--such as combine SYN and FIN flags in case of TCP.

I do agree that a layer on top of `Repr` could be an interesting experiment. I think this layer should cover not just packet construction, but also packet parsing.

I entirely reject your claim that the current API is wrong. It has been designed the way it is and for good reasons.

> I don't think creating new packets and correctly filling them with data is a fairly narrow use case. Like, half the people who use this library are going to want to do that.

I expect most of the people who use this library to never touch the wire module at all, and rather use sockets.

whitequark, 2018-02-08:
@LuoZijun For example, if you already have the payload in memory somewhere, you can use scatter-gather I/O to avoid the need to a copy to prepend headers to data.

canndrew, 2018-02-09:
> Why would you want that? If you want to use scatter-gather capabilities, just use separate buffers for EthernetFrame, Ipv4Packet, UdpPacket, etc. You can do that right now. Splitting the 14-byte Ethernet header over multiple buffers does not appear useful to me.

I have a queue of `EthernetFrame<_>`s that I'm sending out a device. Some of these might be arp, some ipv4, whatever. If I can't just store an entire frame in an `EthernetFrame<_>` the I need to make my own wrapper type that I can keep a queue of - something like:

```rust
type EthernetHeader<T> = EthernetFrame<T>;

enum EthernetPayload<T> {
    Arp(ArpPacket<T>),
    Ipv4(Ipv4Packet<T>),
}

struct ActualEthernetFrame<T> {
    header: EthernetHeader<T>,
    payload: EthernetPayload<T>,
}
```

But that's a lot more complicated and I kinda thought that holding ethernet frames was what the `EthernetFrame` type was for.

Anyway, this is all off-topic for this issue.

> I do agree that a layer on top of Repr could be an interesting experiment. I think this layer should cover not just packet construction, but also packet parsing.
>
> I entirely reject your claim that the current API is wrong. It has been designed the way it is and for good reasons.

To be clear, I'm not saying that current API is *wrong*, just that there could be a much more streamlined and fool-proof API on top of it. Maybe someone wants an API that lets them create invalid packets, but for me that's too general.

So I think we're in agreement. We just need to figure out what that higher-level API could look like. If I have any better ideas I'll post them here.

whitequark, 2018-02-09:
> If I can't just store an entire frame in an EthernetFrame<_> the I need to make my own wrapper type that I can keep a queue of - something like:

This doesn't make any sense to me. The `EthernetFrame` type is not for *storing* anything inside it. It's a read-write wrapper for the ethernet frame *header*. Just like `IpPacket` is a read-write wrapper for the IP packet *header*. These aren't supposed to exist anywhere except stack variables. Once you've set every field you wanted, why would you want to keep the wrapper at all?

If you want to keep a queue of something, make it a queue of `Vec`s or a queue of `Repr`s. Keeping a queue of `EthernetFrame`s is absurd.

canndrew, 2018-02-09:
> Once you've set every field you wanted, why would you want to keep the wrapper at all?

So that I can keep track - at the type level - that a particular buffer contains an ethernet frame.

> If you want to keep a queue of something, make it a queue of `Vec`s

But I don't want a queue of buffers containing arbitrary data, I want a queue of buffers that contain valid ethernet frames. This is what type systems are for - I want my program to be a proof that nothing else can ever make it into my queue.

> or a queue of `Repr`s

`Repr`s don't actually contain the data though. `Ipv4Repr` for instance only contains a `payload_len` field, not the payload itself.

> Keeping a queue of `EthernetFrame`s is absurd.

It's not though. Regardless of whether the library was designed for this sort of usage in mind, it's entirely possible to use it so that `EthernetFrame`/`Ipv4Packet` etc. own their data. It would just be more flexible if that owned data could be split across multiple buffers (although I can understand if you wouldn't want to merge a PR to enable that).

whitequark, 2018-02-09:
> This is what type systems are for - I want my program to be a proof that nothing else can ever make it into my queue.

Yeah, type system tricks like that are an explicit anti-goal of smoltcp. Have you read the README? Second sentence. In my experience, the amount of time you waste preventing you from doing something obviously stupid would be far better spent on something that *doesn't* make your code unreadable and create compiler errors that rival C++ STL.

> Reprs don't actually contain the data though. Ipv4Repr for instance only contains a payload_len field, not the payload itself.

Again, that's because it's a representation of an IPv4 header. The only point of storing `Repr`s in the queue would be because they're more compact, anyway.

> Regardless of whether the library was designed for this sort of usage in mind, it's entirely possible to use it so that EthernetFrame/Ipv4Packet etc. own their data.

This is purely an artifact of Rust being unable to abstract over mutability; the getter impls would not work on `&mut [u8]` if I impl'd them for `&[u8]`. If I could (without making it too onerous), I would make it impossible for them to own data, because that's not what these are for.

> It would just be more flexible if that owned data could be split across multiple buffers (although I can understand if you wouldn't want to merge a PR to enable that).

Did you read my previous responses? You can already have every individual `EthernetFrame`, `IpPacket`, etc, wrap their individual owned buffer. There is nothing to merge.

canndrew, 2018-02-13:
I don't know why you're being so needlessly hostile. Most of your complaints so far make no sense, but I've ignored them because they're irrelevant anyway.

> Yeah, type system tricks like that are an explicit anti-goal of smoltcp. Have you read the README?

Yes I've read the README, and I'm certainly not trying to do any "tricks"

> ... the amount of time you waste preventing you from doing something obviously stupid would be far better spent on something that doesn't make your code unreadable and create compiler errors that rival C++ STL.

Which is more readable? This:

```rust
queue: VecDeque<Bytes>,
```
 or this?

```rust
queue: VecDeque<EthernetFrame<Bytes>>,
```

One says what type of data it's holding (and this is enforced by the compiler!), the other doesn't. I know which one I'd rather encounter when learning a new codebase.

Or how about this?

```rust
if let Some(frame) = queue.pop() {
    let mac_addr = frame.src_addr();
}
```

vs

```rust
if let Some(frame) = queue.pop() {
    let frame = EthernetFrame::new(frame);
    let mac_addr = frame.src_addr();
}
```

The second one just forces me to convert to `EthernetFrame` everywhere I want to use the data, making me write more code and introducing more potential for mistakes.

Nothing I'm doing here is going to make my code produce confusing type errors. If anything I expect the errors to be more useful since the compiler will be able to point more precisely to where I've made the mistake. In my experience this will save me a huge amount of time, both in understanding errors and in preventing them in the first place. But hey, you're allowed to have your own tastes. It's not like I'm trying to merge code like this into smoltcp. I'm just showing you what I'm using it for.

By the way, the main reason that C++ STL errors are so horrifying is due to the lack of a trait/concept system in C++. Instead, C++'s template engine just runs along doing substitutions until it hits an error, at which point it prints out a stack trace of where the error occurred in the form of an extremely long type error. In Rust though, since types have their own type system, this whole process can be avoided and you'll instead just get an error that a type doesn't satisfy a trait.

> If I could (without making it too onerous), I would make it impossible for them to own data, because that's not what these are for.

*What!?* How is "that's not what these are for" a reason to want to break your library for people who are using it like this? If I want to carry a `EthernetFrame<Bytes>` what do you care and why would you want to stop me? The reason I switched to smoltcp is because its types like `Ipv4Packet` are identical to what I'd already coded in my own library (except more complete). So yes, you can use them with owned data. Congrats for writing a library which is more useful than you were imagining.

> Did you read my previous responses? You can already have every individual EthernetFrame, IpPacket, etc, wrap their individual owned buffer. There is nothing to merge.

Yes I did. Did you read mine? I explained how I'm using this library and that having separate buffers like that would be less-than ideal. And I acknowledged that you'd probably be uninterested in merging patches for dis-contiguous storage. Again, I don't understand this attitude. I just want to make smoltcp a better library.

le-jzr, 2018-02-13:
@canndrew You could just define the wrapper type in your user code. I get that your intention is to make smoltcp better according to you own personal preferences, but those don't match the design decisions behind this crate. @whitequark is entirely justified in telling you that you are trying to use the type for something it's not intended for. Yes, you can support all the possible use scenarios one could think of, but that's just bloating the codebase.

whitequark, 2018-02-14:
I don't have any more time for this argument. I'll merge your higher level API for creating packets so long as it doesn't involve explicit allocation and is thus usable in all environments smoltcp is expected to function.

closed 2018-02-14 by whitequark