smoltcp-rs/smoltcp #159: Design: Implementing additional ICMP types

dlrobertson, 2018-02-08:
## Sumary

In order to implement [MLDv2], [NDISC], etc we need to implement additional ICMPv6 types (types that are not defined in [RFC 4443]. There are several reasonable paths forward, each with pros and cons. My plan it to go with [Path 1](#path-1). If there are other ideas or thoughts, please let me know.

## Path1

Create a new module in `wire` for each new protocol with the protocol specific parsing. New variants will need to be added to `Icmpv6Message` along with a generic variant of `Icmpv6Repr` for any message associated with the new protocol. The benefits of this approach include.

  - It will be much more ergonomic to include a separate `process_ndisc` function in `iface`.
  - There will not be as much clutter in the `icmpv6` module.

The drawbacks include

  - The API may look a little different than other protocols in `wire`, due to the message type identifier existing in the ICMPv6 message. This may cause `Packet::new(bytes)` in the new module to be `Packet::new(msg_type, bytes)`.
  - May be less efficient. Depending on the implementation, we might match on `msg_type` more than once to parse the message. Once to identify that it belongs to a "subprotocol", then again to parse the packet according to its type.

## Path 2

The easy path forward would just be to tack on additional variants to the `Icmpv6Repr` enum and add the necessary setters and getters to the `Icmpv6Packet` type. This is by far the simplest of the two, however, it introduces two key drawbacks.

  - There will be some setters and getters for `Icmpv6Packet` that would access fields beyond `HEADER_END`.
  - Eventually there would be a huge number of getters and setters for `Icmpv6Packet` and variants of `Icmpv6Repr`.

[RFC 4443]: https://tools.ietf.org/rfc/rfc4443
[NDISC]: https://tools.ietf.org/rfc/rfc4861
[MLDv2]: https://tools.ietf.org/rfc/rfc3810

whitequark, 2018-02-08:
Definitely path 2, I've already thought about it for ICMPv4 and decided on that. Did you know that you can add documentation to `impl` blocks? So you can have a separate `impl` block per message type.

dlrobertson, 2018-02-08:
> Definitely path 2

It is definitely a reasonable path, especially if the target user is a more advanced user. I'm nervous about users using NDISC or MLDv2 specific packet functions on packets that are not in the family. That is the main reason why I was leaning towards Path 1. But I guess such users probably will not be using the setters and getters in the `Packet` API?

> Did you know that you can add documentation to `impl` blocks?

No I did not. That will definitely come in handy.

whitequark, 2018-02-08:
So, is NDISC a part of ICMPv6 or is it not? It's one of the two.

dlrobertson, 2018-02-08:
NDISC is an extension of ICMPv6. NDISC ([RFC 4861]) defines 5 new ICMPv6 types not defined in [RFC 4443] (the core ICMPv6 RFC). What I'm really referring to is the length difference between the headers for messages defined in [RFC 4443] and some in [RFC 4861]. For example the layout of the Router Advertisement is as follows.

```
      0                   1                   2                   3
      0 1 2 3 4 5 6 7 8 9 0 1 2 3 4 5 6 7 8 9 0 1 2 3 4 5 6 7 8 9 0 1
     +-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
     |     Type      |     Code      |          Checksum             |
     +-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
     | Cur Hop Limit |M|O|  Reserved |       Router Lifetime         |
     +-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
     |                         Reachable Time                        |
     +-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
     |                          Retrans Timer                        |
     +-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
     |   Options ...
     +-+-+-+-+-+-+-+-+-+-+-+-
```

Type etc is the same as the rest of ICMPv6, but `Retrans Timer` etc is well beyond the typical header length of the messages defined in [RFC 4443]. It will make `Icmpv6Packet::check_len` a bit more complicated, but it really shouldn't be too bad.

[RFC 4443]: https://tools.ietf.org/rfc/rfc4443
[RFC 4861]: https://tools.ietf.org/rfc/rfc4861

whitequark, 2018-02-09:
Seems fine to me.

dlrobertson, 2018-05-10:
NDISC has been implemented in `wire`.

closed 2018-05-10 by dlrobertson