smoltcp-rs/smoltcp #160: Design: Implement IPv6 Extension Routing Header

hjr3, 2018-02-10:
The Routing Header spec is at https://tools.ietf.org/html/rfc8200#section-4.4

Two questions about this:

1. What routing types should we implement? IANA routing types are at https://www.iana.org/assignments/ipv6-parameters/ipv6-parameters.xhtml#ipv6-parameters-3.

* Type 0 (aka Source Route) was defined in the original IPv6 RFC but has since been deprecated. Should we implement this route?
* Type 1 (aka `Nimrod`) looks to be abandoned, so that is clear.
* Type 2 is related to IPv6 Mobility https://tools.ietf.org/html/rfc6275#section-6.4. I think this type only makes sense to implement if we implement the rest of [RFC 6275](https://tools.ietf.org/html/rfc6275).
* Type 3 (aka RPL Source Route) is defined in [RFC 6554](https://tools.ietf.org/html/rfc6554) and is the successor to Type 0.
* There are two experiment types defined in [RFC 4727 - Section 3.6](https://tools.ietf.org/html/rfc4727#section-3.6). There is no definitive format specified for these, as they are meant to be used for experiments.

2. How to represent the information for each routing type?

The [format](https://tools.ietf.org/html/rfc8200#section-4.4) of the routing type contains 4 common fields and an unspecified section for _type-specific data_ of variable length.

* Should `Header#check_len` only ensure that there is enough data for the first 4 fields?
* How should we deal with the different methods on each type? For example, there should be a `Header#home_address` method for a Type 3 header, but not one for a Type 2 header.
* How should the `Repr` look? I can imagine something like:

```rust
pub struct Repr<'a> {
    pub next_header: Protocol,
    pub header_len: u8,
    pub routing_type: u8,
    pub segments_left: u8,
    pub type_spec_data: TypeSpecData(&'a [u8])
}

pub enum TypeSpecData<'a> {
    Type2(&'a [u8]),
    Type3(&'a [u8]),
}
```

hjr3, 2018-02-13:
I spoke to @dlrobertson and the goal is to allow parsing of the 4 common fields in the header and ignore the type specific data.

dlrobertson, 2018-02-16:
> What routing types should we implement?

IMO the first round should just implement Type 3.

> How to represent the information for each routing type?

What about an enum repr with struct-like variants (like the ICMPs)

dlrobertson, 2018-05-17:
> Should Header#check_len only ensure that there is enough data for the first 4 fields?

No, it should check to ensure there is enough data for the `Hdr Ext Len` https://tools.ietf.org/html/rfc6554#section-3

> How should we deal with the different methods on each type? For example, there should be a Header#home_address method for a Type 3 header, but not one for a Type 2 header.

See the ICMP packet types. There may be methods that are only valid for specific types. Make sure to add a `Panics` section if the function will panic when used with the wrong type.

> How should the Repr look?

I thing an enum repr with struct like variants is the way to go here.

The Type 3 header has a variable number of addresses that need to be included. To get around having to use `alloc` you might just leave this as a `&'a [u8]` in the `Repr` similar to the ICMP `Repr` for EchoRequests etc. I can't really think of another way around this without requiring `alloc`.

hjr3, 2018-05-22:
Closed by a0009a0b614b71faffb3f0bd3490357e731baaa9

commit a0009a0b (2018-05-22): Add IPv6 Extension Routing Header

Fixes #160

Closes: #218
Approved by: whitequark
closed 2018-05-22 by hjr3