smoltcp-rs/smoltcp #162: Factor wire module out into its own crate?

canndrew, 2018-02-13:
I've written a crate that depends on `smoltcp`, but I'm only using the `wire` module. I just noticed that `smoltcp` requires nightly but I really want to be able to run on stable. The `wire` module alone doesn't require nightly.

Would you be interested in factoring `wire` out into a separate crate, called `smoltcp-wire` or something? I'm probably going to do this anyway. It's really just a matter of who wants to take ownership/responsibility for it and whether we want to de-duplicate code.

dlrobertson, 2018-02-13:
This is duplicate of #54. It may be best close this and post there. Just so everything is in one place.

Either way, thanks for bringing this up! It is actually something I've had a few conversations about recently.

canndrew, 2018-02-13:
Thanks! I'll close this then.

closed 2018-02-13 by canndrew

whitequark, 2018-02-14:
@canndrew Um, what?

First, I am absolutely against you hijacking the `smoltcp` namespace on crates.io. Why did you even think this is okay?

Second, the fact that `smoltcp` depends on nightly Rust is transient. The only nightly feature used is the `slice::rotate_left` function. You're free to wait until that's stabilized (https://github.com/rust-lang/rust/issues/41891), or just inline it into the file where it's used.

canndrew, 2018-02-15:
> First, I am absolutely against you hijacking the smoltcp namespace on crates.io. Why did you even think this is okay?

I didn't. It's a matter of whether you want to create/own the crate or whether I should fork it. If I was to fork it I'd rename it too. I'm sorry if that wasn't clear.

> Second, the fact that `smoltcp` depends on nightly Rust is transient.

Okay, cool. How long do you think until it's stabilised?

whitequark, 2018-02-15:
I have no objections to inlining the function if you want a release of smoltcp that works on stable, like, *right now*.

whitequark, 2018-02-23:
@canndrew smoltcp will work on stable again at 1.26: https://github.com/rust-lang/rust/issues/41891#issuecomment-367862450.