smoltcp-rs/smoltcp #170: Improve socket lookup time.

dlrobertson, 2018-02-24:
The `SocketSet` type is currently a wrapper for a slice causing the complexity of the algorithm to find the socket associated with a given packet to be O(n). We could potentially reduce the complexity and amount of boilerplate code in the `EthernetInterface::process_<protocol>` functions. Perhaps the `SocketSet` should be a wrapper for  a map type.

Note: There is quite a bit that is not specified in this issue that still needs to be thought out.

 - How does `RawSocket` fit?
 - The endpoint type used as keys will have to handle the unspecified address and multiple endpoint types (`IcmpEndpoint` and `IpEndpoint`).
 - `ManagedMap` may not do the trick here, but I haven't done too much research on this.

batonius, 2018-02-24:
I've tried to do that in #19, and even had a working prototype, but eventually lost interest, the `SocketRef` type we have is a remnant of that attempt.

dlrobertson, 2018-02-24:
Gotcha. Didn't look back that far. Post IPv6 I might look into that. Closing this as a duplicate of #19.

closed 2018-02-24 by dlrobertson