smoltcp-rs/smoltcp #174: Add Ipv4Range on wire module ?

LuoZijun, 2018-03-03:
In some case, desciption a network block with CIDR is not simple. like `23.18.1.0  - 23.18.23.255 5888`. 

if we desciption that network block with CIDR, looks like `[ 23.18.1.0/24, 23.18.2.0/23, 23.18.4.0/22, 23.18.8.0/21, 23.18.16.0/21 ]`.

if we have `Ipv4Range` struct, desciption that is easy ( `Ipv4Range(23.18.1.0, 23.18.24.0)` ).

same as python `netaddr` lib: https://github.com/drkjam/netaddr/blob/rel-0.7.x/netaddr/ip/__init__.py#L1422

```rust
extern crate smoltcp;

use smoltcp::wire::{Ipv4Address, Ipv4Cidr};

use std::net::Ipv4Addr;
use std::iter::Iterator;

pub struct Ipv4Range {
    start_ip: Ipv4Address,
    end_ip  : Ipv4Address,
}

impl Ipv4Range {
    pub fn new(start_ip: Ipv4Address, end_ip: Ipv4Address) -> Self {
        Ipv4Range { start_ip, end_ip }
    }
    
    pub fn iter(&self) -> Ipv4RangeIter {
        Ipv4RangeIter {
            start: u32::from(Ipv4Addr::from(self.start_ip.0)),
            end  : u32::from(Ipv4Addr::from(self.end_ip.0)),
        }
    }
}

pub struct Ipv4RangeIter {
    start: u32,
    end  : u32,
}

impl Iterator for Ipv4RangeIter {
    type Item = Ipv4Cidr;

    fn next(&mut self) -> Option<Self::Item> {
        if self.end > self.start {
            let mut shift = self.start.trailing_zeros() as u32;
            let num: u32;
            loop {
                let n = 2u32.pow(shift);
                if self.start + n > self.end {
                    if shift == 0 { panic!("oops ...") }
                    shift -= 1;
                } else {
                    num = n;
                    break;
                }
            }
            let prefix_len = 32 - shift;
            let ip = Ipv4Addr::from(self.start);
            let cidr = Ipv4Cidr::new(Ipv4Address(ip.octets()), prefix_len as u8);
            self.start += num;
            Some(cidr)
        } else if self.end < self.start {
            None
        } else if self.end == self.start {
            None
        } else {
            unreachable!()
        }
    }
}

fn main () {
    // 185.30.232.0 - 185.30.232.63  64
    for cidr in Ipv4Range::new( Ipv4Address::new(185, 30, 232, 0), Ipv4Address::new(185, 30, 232, 64) ).iter() {
        println!("{}", cidr);
    }

    // 23.18.1.0    - 23.18.23.255   5888
    // 23.18.1.0/24
    // 23.18.2.0/23
    // 23.18.4.0/22
    // 23.18.8.0/21
    // 23.18.16.0/21
    for cidr in Ipv4Range::new( Ipv4Address::new(23, 18, 1, 0), Ipv4Address::new(23, 18, 24, 0) ).iter() {
        println!("{}", cidr);
    }
}
```

whitequark, 2018-03-03:
Why should this functionality be in smoltcp? Is it going to be used inside it?

LuoZijun, 2018-03-03:
> Why should this functionality be in smoltcp?

Because ' ipv4range ' and ' IPV4CIDR ' have similar functions.
If this function is isolated, it makes the dependency tree complex.

> Is it going to be used inside it?
I think it should never be.

whitequark, 2018-03-03:
I think the implementation you propose (with `Iterator<Ipv4Cidr>`) is too niche, I don't see a lot of uses for it. Since it can be freely implemented outside of smoltcp I don't think it should be included in smoltcp.

closed 2018-03-03 by LuoZijun