smoltcp-rs/smoltcp #179: Cannot build smoltcp due to an error in 'managed' crate

nisalmenuka2, 2018-03-14:
Hi,
How to solve this issue when building smoltcp? 

`cargo build
   Compiling libc v0.2.39
   Compiling managed v0.5.1
   Compiling bitflags v1.0.1
   Compiling byteorder v1.2.1
   Compiling cfg-if v0.1.2
   Compiling log v0.4.1
error[E0599]: no method named `rotate_left` found for type `[core::option::Option<(K, V)>]` in the current scope
   --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/managed-0.5.1/src/map.rs:161:38
    |
161 |                         pairs[idx..].rotate_left(rotate_by);
    |                                      ^^^^^^^^^^^

error[E0599]: no method named `rotate_left` found for type `[core::option::Option<(K, V)>]` in the current scope
   --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/managed-0.5.1/src/map.rs:187:38
    |
187 |                         pairs[idx..].rotate_left(1);
    |                                      ^^^^^^^^^^^

error: aborting due to 2 previous errors

error: Could not compile `managed`.
warning: build failed, waiting for other jobs to finish...
error: build failed
`

whitequark, 2018-03-14:
You need to update rustc.

nisalmenuka2, 2018-03-14:
Thanks. But can I build it with an older version of Rust? Is it possible to use an older version of the 'managed' crate to overcome the above issue?

whitequark, 2018-03-14:
Not really, since these functions are only available on the nightly channel anyway (they were recently stabilized but are not yet present in a stable release).

closed 2018-05-02 by nisalmenuka2