smoltcp-rs/smoltcp #195: Use new_checked by default

dlrobertson, 2018-04-20:
Currently all `Packet` types have at least the two constructor methods `new` and `new_checked`. `new` does no length checks, while `new_checked` obviously does. AFAIK in `std` etc typically the "safe" version is the default and the "unsafe" version is explicitly called with `<name>_unchecked`. Would it make sense to do something like the following?

 1) deprecate `Packet::new_checked`
 2) make `Packet::new` functions do what `Packet::new_checked` does
 3) add `Packet::new_unchecked` functions that do what `Packet::new` does

dlrobertson, 2018-04-20:
Just a question/thought

whitequark, 2018-04-20:
There is nothing inherently safer about constructing packets with `new_checked` because you can still get panics if you modify a length field inside the packet and then use an accessor.

Moreover, it's not even that one should be preferred over the other. In a typical network stack that both sends and receives packets there would be approximately as many calls to `new` as to `new_checked`.

I'm open to other suggestions for these names, but I think that safety is the wrong angle here.

commit 87ec5fa5 (2018-07-11): Rename `new` method on Packet types to `new_checked`.

Fixes #195.

Closes: #254
Approved by: <try>
commit c13ceef2 (2018-07-11): Rename `new` method on Packet types to `new_checked`.

Fixes #195.

Closes: #254
Approved by: <try>
commit 2a5be8dc (2018-07-11): Rename `new` method on Packet types to `new_checked`.

Fixes #195.

Closes: #254
Approved by: dlrobertson
commit 23acdf57 (2018-07-11): Rename `new` method on Packet types to `new_checked`.

Fixes #195.
closed 2018-07-11 by m-labs-homu