smoltcp-rs/smoltcp #202: socket: accepts function does not need the entire IpRepr

dlrobertson, 2018-05-08:
Each socket type (`ICMP`, `UDP`, and `TCP`) has a corresponding `accetps` function that determines if the given packet was meant for the current socket. Currently this function accepts the entire `IpRepr`, but only the source address and destination address are used.

whitequark, 2018-05-08:
Is this a problem? We already have the `IpRepr`.

dlrobertson, 2018-05-08:
It isn't really a problem. I was primarily thinking of this as unused information. In particular I was thinking about the case where the packet is something like IPv6 > Hop-by-Hop > UDP. In this case the UDP socket `accepts` function will accept the `IpRepr` with a `next_header` value of Hop-by-Hop, but obviously `repr` would be a `UdpRepr`.

whitequark, 2018-05-08:
Yeah, that check is already done in EthernetInterface, so I didn't add it to `accepts` as redundant. We may add it for completeness, although it feels a bit pointless.

Dirbaio, 2020-12-25:
Hello! As part of issue cleanup I'm closing inactive or already-answered "question" issues, since they're not actionable.

Note this doesn't mean questions aren't welcome. They still are! It's just that they shouldn't stay open forever.

closed 2020-12-25 by Dirbaio