smoltcp-rs/smoltcp #209: RFC: multiple routes

ProgVal, 2018-05-14:
As asked in #207, I'm opening an RFC issue about multiple routes.

I created a naive implementation of routes [here](https://github.com/m-labs/smoltcp/compare/master...ProgVal:naive-routes), based on simply iterating over a managed slice.

Of course, there are probably more efficient ways to do it. Any thoughts?

whitequark, 2018-05-14:
Using a managed::ManagedMap would be a first step.

ProgVal, 2018-05-14:
It's not really a map, you would have to iterate through all keys

whitequark, 2018-05-14:
ManagedMap performs binary search on bare no_std and uses a BTreeMap when alloc is available. It never just iterates through every key.

ProgVal, 2018-05-14:
Indeed, sorry.

So what you're thinking of is making `ManagedMap`'s binary-search-related functions public, so we can use it in the route lookup, where we don't need an exact match (as opposed to a map)?

whitequark, 2018-05-14:
No, the API shouldn't deviate that much. `ManagedMap` needs to gain BTreeMap's `range` method.

ProgVal, 2018-06-09:
Closed by #219

closed 2018-06-09 by ProgVal