smoltcp-rs/smoltcp #211: RFC: Adding support for repeated NDISC options

ProgVal, 2018-05-14:
I noticed some code duplication in `NdiscOptions`'s handling: each variant of `Message` has to do its own loop over options, to find `SourceLinkLayerAddr` and possibly other options. `Message::Redirect` handling also had code that also exists `ndiscoption.rs`.

So I factorized all this, by adding an iterator structure over NDISC options, with enough room to store up to 32 options. My guess is that it should be enough, but I am not sure (Only `Prefix information` should be used more than once. Do you know if there are routers that announce more than 31 prefixes at the same time?)
[Here is a diff](https://github.com/m-labs/smoltcp/compare/master...ProgVal:ndisc-options)

I also tried to implement that iterator with a `ManagedSlice`, but that requires to pass a pre-allocated slice through `EthernetInterface`'s API, and I find that messy.

whitequark, 2018-05-14:
Why does the iterator need to use a fixed size slice? This seems quite ugly to me.

ProgVal, 2018-05-14:
Sorry, I forgot to mention this. I can't allocate on the heap without requiring `alloc` or `std`.

whitequark, 2018-05-14:
Yeah, but can't you borrow a slice?

ProgVal, 2018-05-14:
It works to create a `NdiscOptions` or a `NdiscOptionsRepr` from scratch, but not with `NdiscOptionsRepr::parse`, because we need another slice, and I don't know where to get it from

dlrobertson, 2018-05-14:
Or a macro could work too.

dlrobertson, 2018-05-14:
The real downside with flattening the options into the `NdiscRepr` like we currently do is `) it causes us to only know the options included in the `Repr` and 2) we keep duplicates. E.g. a Router Advertisement with two Prefix Information options will only result in a `Repr` with one Prefix Information option.

ProgVal, 2018-05-20:
We could keep duplicates, but only a fixed number of them, known at compile time. Whereas in the RFC, the only limit is the MTU

whitequark, 2018-05-24:
Do I understand it correctly that right now we handle any number of options just fine on no_std, but with this refactoring, we will be limited to something that is less than what RFC allows? This seems clearly bad to me.

ProgVal, 2018-05-24:
No, right now, only one Prefix Information option is handled.

whitequark, 2018-05-24:
OK, I'll need to take a closer look at this.

dlrobertson, 2018-05-25:
> we will be limited to something that is less than what RFC allows

Yes the RFC allows for multiple options, but we do not currently support it.

> There can also be multiple instances of the same option in a message (e.g., Prefix Information options). [1]

[1]: https://tools.ietf.org/html/rfc4861#section-9

dlrobertson, 2018-11-01:
After spending some time reviewing the linux kernel, it seems that the linux kernel does not allow for repeated NDISC options.

```c
struct ndisc_options *ndisc_parse_options(const struct net_device *dev,
                                          u8 *opt, int opt_len,
                                          struct ndisc_options *ndopts)
{
        ...
                switch (nd_opt->nd_opt_type) {
                case ND_OPT_SOURCE_LL_ADDR:
                case ND_OPT_TARGET_LL_ADDR:
                case ND_OPT_MTU:
                case ND_OPT_NONCE:
                case ND_OPT_REDIRECT_HDR:
                        if (ndopts->nd_opt_array[nd_opt->nd_opt_type]) {
                                ND_PRINTK(2, warn,
                                          "%s: duplicated ND6 option found: type=%d\n",
                                          __func__, nd_opt->nd_opt_type);
                        } else {
                                ndopts->nd_opt_array[nd_opt->nd_opt_type] = nd_opt;
                        }
                        break;
```

So I don't think it is unreasonable that we also do not allow duplicate options.

whitequark, 2018-11-07:
@dlrobertson Doing the same thing as Linux seems good to me.

closed 2022-06-06 by Dirbaio