smoltcp-rs/smoltcp #222: [RFC]: Break out the core modules into cargo workspaces

dlrobertson, 2018-05-29:
## Summary

The core modules in `smoltcp` (especially `wire`) could provide value as individual crates. The `[workspace]` feature in cargo should allow us to break out each module into a separate crate without changing the current dev workflow too much.

## Detailed Description

Each core module (`wire`, `phy`, `iface`, etc.) will be relocated to the root, a manifest will be added, and an entry for the directory will be added to the roots [Virtual Manifest](https://doc.rust-lang.org/cargo/reference/manifest.html#virtual-manifest). The end result should be several crates made available (e.g. `smolwire`, `smolphy`, and `smoliface`).

## Open Questions

 - Is it worth it?
   - I think so and I'm willing to work on this, but this has an impact on the dev workflow, and will likely cause conflicts in open PRs. As a result, this should not be taken lightly.
 - Is the cargo workspace and monolithic repo the right approach?
 - Are there examples of other crates/groups that currently use `[workspace]` to solve a similar problem? Has their experience been positive?
 - I think `cargo publish` should still work with `cargo publish --all`, but this should be verified.
 - What should the naming scheme be?

## References

[Cargo `workspace`](https://doc.rust-lang.org/cargo/reference/manifest.html#the-workspace-section)

whitequark, 2018-05-29:
I personally don't think this is worth it. The compilation time of smoltcp is less than a second last time I checked, there is no global data involved, and all functionality that can bloat unrelated codepaths is gated with feature flags.

What do we gain exactly?

dlrobertson, 2018-05-29:
`iface` is compiled regardless. Yes you can disable all the socket types, but IIRC `Interface` etc is still compiled in.

> The compilation time of smoltcp is less than a second

AFAIK this shouldn't really affect compile time too much.

> Whenever any crate in the workspace is compiled, output is placed in the workspace root. i.e. next to the root crate's Cargo.toml. [1]

My cargo knowledge is weak at best, but if all the workspaces are using the root `target` directory and they share a `Cargo.lock`, wouldn't that be similar to when they're modules. It wouldn't be exactly the same, but I think it would be close. I could do some benchmarks if you'd like.

> What do we gain exactly?

Quite a few people have requested it. Primarily a `wire` seems very valuable. If `wire` isn't broken out into a separate crate perhaps adding a section to the `README.md` about how to compile smoltcp as a packet parsing framework would be good.

[1]: https://doc.rust-lang.org/cargo/reference/manifest.html#the-workspace-section

dlrobertson, 2018-05-29:
Primarily your comment [here](https://github.com/m-labs/smoltcp/issues/54#issuecomment-339172433) made me think this could be a possibility now.

> Right now I don't really have the bandwidth to review everything, but I don't see why we can't do this a bit later when we have more contributors.

dlrobertson, 2018-06-07:
> In any case, I like the idea of adding a feature for turning off EthernetInterface far more than splitting into many crates.

I suppose someone who needs `wire` will also need `phy` for `ChecksumCapabiliites` anyways, so rather than requiring two separate crates, it would be easier to just use one crate without a specific feature enabled.

Dirbaio, 2022-06-06:
Closing as wontfix.

The goal of Cargo features is to avoid code size bloat due to unused enum variants in an used enum, which the compiler can't see are unused.

If you don't use anything from `iface` at all, the compiler already doesn't include any code from it, so this wouldn't gain code size.

The other reason for features is compile speed, but smoltcp already compiles very fast so it's not worth it IMO.

closed 2022-06-06 by Dirbaio