smoltcp-rs/smoltcp #223: Unable to create UDP and TCP sockets

crawford, 2018-05-31:
I'm trying to create two listening sockets, one UDP and one TCP, with the following code:

```rust
    let mut udp_rx_payload = [0; 128];   
    let mut udp_rx_metadata = [PacketMetadata::EMPTY; 1];
    let mut udp_tx_payload = [0; 128];
    let mut udp_tx_metadata = [PacketMetadata::EMPTY; 1];
    let udp_socket = UdpSocket::new(
        UdpSocketBuffer::new(udp_rx_metadata.as_mut(), udp_rx_payload.as_mut()),
        UdpSocketBuffer::new(udp_tx_metadata.as_mut(), udp_tx_payload.as_mut()),
    );                        

    let mut tcp_rx_payload = [0; 128];                      
    let mut tcp_tx_payload = [0; 128];
    let tcp_socket = TcpSocket::new(
        TcpSocketBuffer::new(tcp_rx_payload.as_mut()),                         
        TcpSocketBuffer::new(tcp_tx_payload.as_mut()),
    );
            
    let mut sockets = [None, None];
    let mut socket_set = SocketSet::new(sockets.as_mut());
    let udp_handle = socket_set.add(udp_socket);
    let tcp_handle = socket_set.add(tcp_socket);
```

When I try to build this, I get the following error:

```
error[E0597]: `udp_rx_payload` does not live long enough
   --> src/main.rs:117:56
    |
117 |         UdpSocketBuffer::new(udp_rx_metadata.as_mut(), udp_rx_payload.as_mut()),
    |                                                        ^^^^^^^^^^^^^^ borrowed value does not live long enough
...
167 | }
    | - borrowed value only lives until here
    |
    = note: borrowed value must be valid for the static lifetime...

error[E0597]: `udp_tx_payload` does not live long enough
   --> src/main.rs:118:56
    |
118 |         UdpSocketBuffer::new(udp_tx_metadata.as_mut(), udp_tx_payload.as_mut()),
    |                                                        ^^^^^^^^^^^^^^ borrowed value does not live long enough
...
167 | }
    | - borrowed value only lives until here
    |
    = note: borrowed value must be valid for the static lifetime...
```

If I comment out either of the calls to `socket_set.add`, it builds and runs.

I'm using rustc 1.28.0-nightly (c2d46037f 2018-05-24).

whitequark, 2018-05-31:
The lifetimes of all sockets unify with each other, so you need to group them like this:
```rust
    let mut udp_rx_payload = [0; 128];   
    let mut udp_tx_payload = [0; 128];
    let mut tcp_rx_payload = [0; 128];                      
    let mut tcp_tx_payload = [0; 128];
    let mut udp_tx_metadata = [PacketMetadata::EMPTY; 1];
    let mut udp_rx_metadata = [PacketMetadata::EMPTY; 1];
    let udp_socket = UdpSocket::new(
        UdpSocketBuffer::new(udp_rx_metadata.as_mut(), udp_rx_payload.as_mut()),
        UdpSocketBuffer::new(udp_tx_metadata.as_mut(), udp_tx_payload.as_mut()),
    );                        
    let tcp_socket = TcpSocket::new(
        TcpSocketBuffer::new(tcp_rx_payload.as_mut()),                         
        TcpSocketBuffer::new(tcp_tx_payload.as_mut()),
    );
            
    let mut sockets = [None, None];
    let mut socket_set = SocketSet::new(sockets.as_mut());
    let udp_handle = socket_set.add(udp_socket);
    let tcp_handle = socket_set.add(tcp_socket);
```
cf. #48

crawford, 2018-05-31:
That gives me the same error as before:

```
error[E0597]: `udp_rx_payload` does not live long enough
   --> src/main.rs:119:56
    |
119 |         UdpSocketBuffer::new(udp_rx_metadata.as_mut(), udp_rx_payload.as_mut()),
    |                                                        ^^^^^^^^^^^^^^ borrowed value does not live long enough
...
166 | }
    | - borrowed value only lives until here
    |
    = note: borrowed value must be valid for the static lifetime...

error[E0597]: `udp_tx_payload` does not live long enough
   --> src/main.rs:120:56
    |
120 |         UdpSocketBuffer::new(udp_tx_metadata.as_mut(), udp_tx_payload.as_mut()),
    |                                                        ^^^^^^^^^^^^^^ borrowed value does not live long enough
...
166 | }
    | - borrowed value only lives until here
    |
    = note: borrowed value must be valid for the static lifetime...
```

Is there something in the surrounding code that I need to check? All of this code exists in `main`.

whitequark, 2018-05-31:
Oh sorry, I didn't go far enough. You need...
```rust
    let mut udp_rx_payload = [0; 128];   
    let mut udp_tx_payload = [0; 128];
    let mut tcp_rx_payload = [0; 128];                      
    let mut tcp_tx_payload = [0; 128];
    let mut udp_tx_metadata = [PacketMetadata::EMPTY; 1];
    let mut udp_rx_metadata = [PacketMetadata::EMPTY; 1];
    let udp_rx_buffer = UdpSocketBuffer::new(udp_rx_metadata.as_mut(), udp_rx_payload.as_mut());
    let udp_tx_buffer = UdpSocketBuffer::new(udp_tx_metadata.as_mut(), udp_tx_payload.as_mut());
    let tcp_rx_buffer = TcpSocketBuffer::new(tcp_rx_metadata.as_mut(), tcp_rx_payload.as_mut());
    let tcp_tx_buffer = TcpSocketBuffer::new(tcp_tx_metadata.as_mut(), tcp_tx_payload.as_mut());
    let udp_socket = UdpSocket::new(udp_rx_buffer, udp_tx_buffer);                        
    let tcp_socket = TcpSocket::new(tcp_rx_buffer, tcp_tx_buffer);
            
    let mut sockets = [None, None];
    let mut socket_set = SocketSet::new(sockets.as_mut());
    let udp_handle = socket_set.add(udp_socket);
    let tcp_handle = socket_set.add(tcp_socket);
```

crawford, 2018-05-31:
(There is a small typo above. `TcpSocketBuffer::new` doesn't need metadata). I still get the same error :(

whitequark, 2018-05-31:
Can you post your project somewhere? In any case, here's a working example: https://github.com/m-labs/ionpak/blob/master/firmware/src/main.rs#L158-L194.

crawford, 2018-05-31:
I just made it public. Here is the code in question (without the suggested changes): https://github.com/crawford/PoE/blob/95d11340f3e2025f3930e5bd4fa4108acaf96968/firmware/src/main.rs#L112.

ProgVal, 2018-06-21:
@crawford Hmm, I can compile your code with the nightly compiler. (Maybe thanks to the NLL?)

crawford, 2018-07-03:
@ProgVal Did you uncomment line 131?

crawford, 2018-07-03:
@whitequark I'm now getting an eerily similar error after (refactoring my Device implementation](https://github.com/crawford/PoE/blob/78817d4dbcb2d0a1bc981238a184996b5029b3c2/firmware/src/main.rs#L109). This code fails with the following error message:

```
error[E0597]: `tx_buffer` does not live long enough
   --> src/main.rs:111:18
    |
111 |             &mut tx_buffer,
    |                  ^^^^^^^^^ borrowed value does not live long enough
...
178 | }
    | - `tx_buffer` dropped here while still borrowed
    |
    = note: values in a scope are dropped in the opposite order they are created
```

I wonder if both this failure and the original actually stem from the type definition of Interface and InterfaceBuilder. Both of these structs dictate that DeviceT is a Device of any lifetime (including 'static), but my Device does not implement 'static (unlike the Device in IonPak). I've been trying to modify smoltcp to lift that restriction, but I haven't been able to satisfy the borrow checker yet.

birkenfeld, 2018-07-03:
Does it change anything if you move `rx_buffer`/`tx_buffer` above the creation of `neighbor_cache` and `ip_addrs`?

crawford, 2018-07-03:
@birkenfeld Unfortunately, no.

crawford, 2019-08-18:
This has been fixed by https://github.com/m-labs/smoltcp/pull/304.

closed 2019-08-18 by crawford