smoltcp-rs/smoltcp #224: Debug build panic local_rx_dup_acks overflow

pothos, 2018-06-01:
This code here https://github.com/m-labs/smoltcp/blob/master/src/socket/tcp.rs#L913 should either reset `local_rx_dup_acks` or let it count from 255 to 0 with a overflowing add or stay at 255 with a saturation. What is the desired behavior?
```
                        self.local_rx_dup_acks += 1;
                        if self.local_rx_dup_acks == 3 {
                        }

```
`thread 'main' panicked at 'attempt to add with overflow', smoltcp/src/socket/tcp.rs:913:25`

pothos, 2018-06-01:
Actually master stopped working for me, I try to find out if related to the fast retrainsmit code.

whitequark, 2018-06-01:
cc @barskern

barskern, 2018-06-01:
@pothos Do you think you would be able to write a test which mimics this behavior? Or explain me on how this managed to happen so that I could write a test?

I would think that the desired behavior is to do a saturating addition, and change the condition from `if self.local_rx_dup_acks == 3` to `if self.local_rx_dup_acks >= 3`. This would trigger a fast retransmit for every segment which is a duplicate ACK when we have recived more than 3 duplicate ACKs. However I did not find documentation that indicates that this is appropriate behavior for a TCP-implementation.

barskern, 2018-06-02:
> Per my understanding, with the modified code, if e.g. first packet in a burst of 8 gets lost, the other 7 will arrive, and 4 of them will trigger retransmit of the entire burst. Is that correct?

This is the reason I previously used `== 3`. However if we recive a burst of pacakges and the socket is not polled until all 8 packages have recived, I think there will only be 1 actual retransmit. This is because the way the socket retransmits is by setting an immidiate timeout, not directly sending a packet. Or thats what I think.. Perhaps we could find a way to test this theory.

This is such a edge case though. What is the probablility that we lose a packet and then lose the exact same package again in the retransmit, when all the other packages arrived. I think this is why I can't find a document about it online.

barskern, 2018-06-02:
After some more research, I found that what should happen after a duplicate ACK is recived is that the socket should enter "Fast recovery" and every following duplicate ACK only increases the congestion window. 

> After the fast retransmit algorithm sends what appears to be the
   missing segment, the "fast recovery" algorithm governs the
   transmission of new data until a non-duplicate ACK arrives. [RFC 5681, page 9](https://tools.ietf.org/html/rfc5681#page-9).

Hence since there hasn't been implemented Congestion Control in this library, I think we just have to "ignore" further duplicate ACKs and wait for timeout to retransmit again.

pothos, 2018-06-04:
Ah, I applied the proposed patch with a saturating add, btw. So the numbers jumping back again should not be caused by overflows.

barskern, 2018-06-04:
@pothos I found a bug in the fast retransmit code based on your logs! 

When there is content in a package, one should not count the ACK as a duplicate ACK! New PR incoming 👍

pothos, 2018-06-05:
@barskern Yes, that's one point, but also the retransmit should only be scheduled if there is data to be sent out. In the tcpdump in the end the packets of both sides just have length 0.

whitequark, 2018-06-06:
Thanks for debugging this!

barskern, 2018-06-09:
@pothos Could you run the same test that generated this issue in the first place? I'm excited to see if it works now :zap:

pothos, 2018-06-12:
Hi,
sorry, but no, two endpoints can still be stuck sending the same packets.

barskern, 2018-06-12:
Could you upload debug logs again?

pothos, 2018-06-12:
I've found the issues and fixed it but did not write a test yet: https://github.com/m-labs/smoltcp/pull/233

pothos, 2018-06-18:
Hi,
can somebody review the PR to get master working again?
Thanks

closed 2018-06-19 by pothos