smoltcp-rs/smoltcp #227: Server example broken

podhrmic, 2018-06-05:
Commit https://github.com/m-labs/smoltcp/commit/2363f97834b3d529c6344f1840ab001645e853d2 broke server example:

```
$ cargo run --example server tap1
warning: unused manifest key: package.autoexamples
   Compiling smoltcp v0.4.0 (file:///home/michal/Workspace/rust_projects/smoltcp)
    Finished dev [unoptimized + debuginfo] target(s) in 5.66 secs
     Running `target/debug/examples/server tap1`
[1528220661.171s] (socket::set): [0]: adding
[1528220661.171s] (socket::set): [1]: adding
[1528220661.171s] (socket::set): [2]: adding
[1528220661.171s] (socket::set): [3]: adding
[1528220661.172s] (socket::set): [4]: adding
[1528220661.172s] (socket::tcp): #1:*:6969: state=CLOSED=>LISTEN
[1528220661.172s] (socket::tcp): #2:*:6970: state=CLOSED=>LISTEN
[1528220661.172s] (socket::tcp): #3:*:6971: state=CLOSED=>LISTEN
[1528220661.172s] (socket::tcp): #4:*:6972: state=CLOSED=>LISTEN
[1528220661.183s] <- EthernetII src=5e-13-5d-a6-c5-f2 dst=33-33-00-00-00-16 type=IPv6
                \ IPv6 src=fe80::5c13:5dff:fea6:c5f2 dst=ff02::16 nxt_hdr=Hop-by-Hop hop_limit=1
[1528220661.183s] (iface::neighbor): filled fe80::5c13:5dff:fea6:c5f2 => 5e-13-5d-a6-c5-f2 (was empty)
[1528220661.183s] (iface::ethernet): cannot process ingress packet: unrecognized packet
[1528220661.183s] (iface::ethernet): packet dump follows:
EthernetII src=5e-13-5d-a6-c5-f2 dst=33-33-00-00-00-16 type=IPv6
\ IPv6 src=fe80::5c13:5dff:fea6:c5f2 dst=ff02::16 nxt_hdr=Hop-by-Hop hop_limit=1
thread 'main' panicked at 'poll error: Unrecognized', libcore/result.rs:945:5
note: Run with `RUST_BACKTRACE=1` for a backtrace.
```

The correct behavior (with commit https://github.com/m-labs/smoltcp/commit/dd80b22a98fd60da9602779295b073def3b9ecf3 ) is:

```
$ cargo run --example server tap1
warning: unused manifest key: package.autoexamples
   Compiling smoltcp v0.4.0 (file:///home/michal/Workspace/rust_projects/smoltcp)
    Finished dev [unoptimized + debuginfo] target(s) in 5.76 secs
     Running `target/debug/examples/server tap1`
[1528220498.560s] (socket::set): [0]: adding
[1528220498.560s] (socket::set): [1]: adding
[1528220498.560s] (socket::set): [2]: adding
[1528220498.560s] (socket::set): [3]: adding
[1528220498.560s] (socket::set): [4]: adding
[1528220498.560s] (socket::tcp): #1:*:6969: state=CLOSED=>LISTEN
[1528220498.560s] (socket::tcp): #2:*:6970: state=CLOSED=>LISTEN
[1528220498.560s] (socket::tcp): #3:*:6971: state=CLOSED=>LISTEN
[1528220498.560s] (socket::tcp): #4:*:6972: state=CLOSED=>LISTEN
[1528220498.571s] <- EthernetII src=5e-13-5d-a6-c5-f2 dst=33-33-00-00-00-16 type=IPv6
                \ IPv6 src=fe80::5c13:5dff:fea6:c5f2 dst=ff02::16 nxt_hdr=Hop-by-Hop hop_limit=1
[1528220498.571s] (iface::neighbor): filled fe80::5c13:5dff:fea6:c5f2 => 5e-13-5d-a6-c5-f2 (was empty)
[1528220498.667s] <- EthernetII src=5e-13-5d-a6-c5-f2 dst=33-33-00-00-00-16 type=IPv6
                \ IPv6 src=fe80::5c13:5dff:fea6:c5f2 dst=ff02::16 nxt_hdr=Hop-by-Hop hop_limit=1
```

dlrobertson, 2018-06-05:
This is due to the current lack of MLDv2 support (currently working on it). The reason dd80b22 doesn't return `Unrecognized` is because we didn't continue processing a packet that had `Hop-by-Hop` options present.

dlrobertson, 2018-06-05:
IMO this isn't a bug. We can either 1) wait until MLDv2 support is finished or 2) update the server example to not panic on `Unrecognized`.

podhrmic, 2018-06-05:
In that case, what do you think about replacing `iface.poll(&mut sockets, timestamp).expect("poll error");` in server with proper match statement, so the errors are printed/logged, but the server example keeps running? It just seems to be silly to not panic on `Unrecognized` but fail otherwise.

dlrobertson, 2018-06-05:
Per the [poll documentation], I think logging an error is reasonable.

[poll documentation]: https://github.com/m-labs/smoltcp/blob/master/src/iface/ethernet.rs#L359-L368

closed 2018-06-06 by dlrobertson