smoltcp-rs/smoltcp #242: Fix clippy warnings

dlrobertson, 2018-06-19:
The following clippy lints throw errors or warnings. We should either explicitly allow them or fix the code that fails the lint.

### Errors
 - [ ] [if_same_then_else](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#if_same_then_else)

### Warnings
 - [ ] [assign_op_pattern](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#assign_op_pattern)
 - [ ] [cast_lossless](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#cast_lossless)
 - [ ] [clone_on_copy](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#clone_on_copy)
 - [ ] [collapsible_if](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#collapsible_if)
 - [ ] [cyclomatic_complexity](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#cyclomatic_complexity)
 - [ ] [duration_subsec](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#duration_subsec)
 - [ ] [explicit_iter_loop](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#explicit_iter_loop)
 - [ ] [identity_op](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#identity_op)
 - [ ] [large_enum_variant](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#large_enum_variant)
 - [ ] [len_without_is_empty](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#len_without_is_empty)
 - [ ] [len_zero](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#len_zero)
 - [ ] [let_and_return](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#let_and_return)
 - [ ] [match_bool](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#match_bool)
 - [ ] [match_ref_pats](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#match_ref_pats)
 - [ ] [needless_lifetimes](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#needless_lifetimes)
 - [ ] [needless_pass_by_value](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#needless_pass_by_value)
 - [ ] [needless_range_loop](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#needless_range_loop)
 - [ ] [needless_return](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#needless_return)
 - [ ] [new_without_default_derive](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#new_without_default_derive)
 - [ ] [redundant_closure](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#redundant_closure)
 - [ ] [redundant_field_names](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#redundant_field_names)
 - [ ] [search_is_some](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#search_is_some)
 - [ ] [single_match](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#single_match)
 - [ ] [too_many_arguments](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#too_many_arguments)
 - [ ] [trivially_copy_pass_by_ref](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#trivially_copy_pass_by_ref)
 - [ ] [unnecessary_mut_passed](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#unnecessary_mut_passed)
 - [ ] [unreadable_literal](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#unreadable_literal)
 - [ ] [useless_asref](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#useless_asref)
 - [ ] [useless_let_if_seq](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#useless_let_if_seq)
 - [ ] [write_with_newline](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#write_with_newline)
 - [ ] [wrong_self_convention](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#wrong_self_convention)

whitequark, 2018-06-19:
I'm not extremely enthusiastic about clippy and I believe that blindly following its suggestions can easily degrade code quality. r=me on every PR addressing its lints.

dlrobertson, 2018-06-19:
>  believe that blindly following its suggestions can easily degrade code quality.

I agree 100% with this. I'm definitely __not__ suggesting we blindly follow the suggestions. [unreadable_literal](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#unreadable_literal) is a great example of a lint that IMO doesn't help. IMO `0xffff_ffff` does not help readability at all. I do think that we could improve the code quality by following some of these e.g. [unnecessary_mut_passed](https://rust-lang-nursery.github.io/rust-clippy/v0.0.208/index.html#unnecessary_mut_passed).

> r=me on every PR addressing its lints.

Will do.

jhwgh1968, 2019-05-05:
After doing a nightly compiler update, I noticed this new warning:

```
warning[E0713]: borrow may still be in use when destructor runs
  --> src/socket/ref_.rs:61:9
   |
39 | impl<'a, T: Session + 'a> Ref<'a, T> {
   |      -- lifetime `'a` defined here
...
61 |         ref_.socket
   |         ^^^^^^^^^^^ returning this value requires that `*ref_.socket` is borrowed for `'a`
62 |     }
   |     - here, drop of `ref_` needs exclusive access to `*ref_.socket`, because the type `socket::ref_::Ref<'_, T>` implements the `Drop` trait
   |
   = warning: this error has been downgraded to a warning for backwards compatibility with previous releases
   = warning: this represents potential undefined behavior in your code and this warning will become a hard error in the future
```

Since I presume the code not causing any problems right now, I thought it should be addressed along with the clippy lints. If this should be a separate issue, I can easily create one.

Dirbaio, 2021-01-06:
This is now done. Additionally, CI enforces no clippy warnings.

closed 2021-01-06 by Dirbaio