smoltcp-rs/smoltcp #245: strange behavior with packet buffer enqueue?

birkenfeld, 2018-06-20:
I'm receiving two alternating messages (`start` and `stop`) on a UDP socket. The third cannot be processed. I added some logging to `PacketBuffer::enqueue`, seen in the debug output here:
```
[DEBUG] full: false, window: 6000, contig_window: 6000, size: 5
[DEBUG] full: false, window: 6000, contig_window: 5995, size: 4
[DEBUG] full: false, window: 6000, contig_window: 5996, size: 5
[DEBUG] cannot process ingress packet: buffer space exhausted
```
The third packet triggers the condition here:
```
        if window < size || (window != contig_window && window - contig_window < size) {
            return Err(Error::Exhausted)
        }
```
because `window - contig_window == 4 < size == 5`. But as far as I understand, having a contiguous window larger than `size` cannot be an exhaustion condition?

whitequark, 2018-06-20:
Which version of smoltcp?

whitequark, 2018-06-20:
@dlrobertson Can you please take a look? Something's screwy with that condition.

dlrobertson, 2018-06-20:
Yup, that logic seems off...

birkenfeld, 2018-06-21:
FWIW, it's revision 216c393 (the switch to `managed` 0.7 breaks with my pegged nightly).

dlrobertson, 2018-06-21:
@birkenfeld thanks for pointing this out. The linked PR fixes this.

closed 2018-06-21 by dlrobertson