smoltcp-rs/smoltcp #246: wrong assertion in packet_buffer tests

birkenfeld, 2018-06-21:
master, `src/storage/packet_buffer.rs` has this test:
```
    fn buffer() -> PacketBuffer<'static, 'static, ()> {
        PacketBuffer::new(vec![PacketMetadata::EMPTY; 4],
                          vec![0u8; 16])
    }
    #[test]
    fn test_padding() {
        let mut buffer = buffer();
        assert!(buffer.enqueue(6, ()).is_ok());
        assert!(buffer.enqueue(8, ()).is_ok());
        assert!(buffer.dequeue().is_ok());
        buffer.enqueue(4, ()).unwrap().copy_from_slice(b"abcd");
        assert_eq!(buffer.metadata_ring.len(), 3);
        ...
    }
```
Now after enqueueing three times and dequeueing once, all successfully, shouldn't the number of elements in the metadata ring be 2 instead of 3?

Even stranger, when I increase the size of the payload buffer from 16 to say 32, this test (and `test_dequeue_with` below, which has the same assertion) fails.

whitequark, 2018-06-21:
No, this is not wrong. There is a special element in the metadata ring that means "there is a packet after this element, but this element itself does not signify anything, it only consumes space in the tail end of the buffer because this space was not sufficient to insert the next packet". Aka "padding".

closed 2018-06-21 by whitequark

birkenfeld, 2018-06-21:
Ah, thanks, makes sense.