smoltcp-rs/smoltcp #256: Implement TCP selective acknowledgement extension (sACK)

jhwgh1968, 2018-07-22:
The TCP extension is described in [RFC 2018](https://tools.ietf.org/html/rfc2018).

I am opening a tracking issue for this since I didn't see one. If no one else wants to pick this up, I volunteer.

@whitequark previously made some offhanded comments suggesting it would be easy to send sACK records in acknowledgements based on current gaps tracked by the TCP reassembly code. I tend to agree, based on my understanding of the code.

The more difficult part is actually using information in sACK records sent by the remote TCP. While completely ignoring the remote's sACK records is legal -- the ACK field must not advance past the first gap to prevent mismatching state -- it defeats the extension's purpose, and so is highly discouraged.

whitequark, 2018-09-24:
> It seems when I write a straightforward case (do a bunch of `send!` calls with frames but one missing from the sequence), the `recv!` call at the end -- expecting an ACK with sACK in the TCP options -- gets `Ok(None)`, which causes an assert failure. And I cannot figure out why.

Where is your SACK code located? Is it just in `TcpSocket::process` or are you also emitting SACKs in `TcpSocket::dispatch`?

whitequark, 2018-09-28:
@jhwgh1968 You can't call `s.rx_buffer.dequeue*` if you want to continue using the socket afterwards. Take a look at the implementation of `TcpSocket::recv`; you're skipping `TcpSocket::recv_impl`, which means `s.remote_seq_no` doesn't get updated, and the assembler gets confused afterwards.

I think it would be ideal if we replaced all uses of `s.rx_buffer.dequeue*` in tests with `s.recv*`.

jhwgh1968, 2018-10-09:
I have updated my test branch, and replaced the ``dequeue`` calls with ``recv``. It seems to still return ``Ok(None)``.

What else am I missing?

whitequark, 2018-11-08:
Great to hear, and thanks for your work and patience!

jhwgh1968, 2019-01-01:
Since it didn't get linked to this issue, I have a phase 1 implementation in #266.

I will probably work on a phase 1.5 next, where it actually becomes fully "should" compliant with its receiving behavior.

jhwgh1968, 2019-03-11:
Unfortunately, I have not had time to work on this. If anyone else wants to take this issue, feel free. Otherwise, I may get to it later in the year.

jabedude, 2020-05-21:
@jhwgh1968 what is left to implement?

jhwgh1968, 2020-05-22:
I'm glad you asked, @jabedude! I tried to finish this at one point during the year, and didn't get very far.

The way I planned it in my mind, there will be several phases. As you will see, the further into the future the plan is, the more vague it is.

### Sender Side Phase 1

This is what I have done already. To quote myself:

> This PR correctly handles all "must" requirements of RFC 2018:
> 
>  *  The sACK Permitted and sACK Range options are transmitted in accordance with the RFC.
>  *  If the remote does not send sACK permitted, sACK range options are never transmitted.
>  *  When a sACK range is transmitted, it contains the range of data corresponding to the segment that triggered the ACK unless the segment caused the ACK field to update (i.e. filled in an assembler hole that permitted progress).
>  *  Once a sACK range is announced, that data is never thrown away. (This is not a "must" requirement, but if the TCP does this, then additional "must" requirements are added.)
> 
> However, since this is a "phase 1" PR, compliance is all this implementation has to offer. In practical terms, it only improves throughput in one case, and even then, only for data flowing in one direction.
> 
> This is because of the shortcuts taken to save significant complexity:
> 
>  *  All sACK information received from the remote side is ignored
>  *  Only one sACK range will ever be transmitted, even though the data structures support up to 3
>  *  If the segment causes an update in the ACK field, smoltcp will choose a "helpful hint" range if there are still holes in the assembler. The algorithm it uses to select this is contrary to a "should" in the RFC.
> 
> Each one of these is avoiding a significant effort. The first is a whole new data structure, the second two are because the "should" algorithm is based on keeping a history of transmitted sACK ranges on a queue, and pulling them off as holes are filled.

### Sender Side Phase 2

This finishes the sender side implementation to be "should" compliant with the RFC.

In particular, RFC case 3 should be turned into a test. (In addition to several other interesting edge case tests, of course.)

<details>
  Case 3:  The 2nd, 4th, 6th, and 8th (last) segments are
  dropped.

  The data receiver ACKs the first packet normally.  The
  third, fifth, and seventh packets trigger SACK options as
  follows:

    Triggering  ACK    First Block   2nd Block     3rd Block
    Segment            Left   Right  Left   Right  Left   Right
                       Edge   Edge   Edge   Edge   Edge   Edge

    5000       5500
    5500       (lost)
    6000       5500    6000   6500
    6500       (lost)
    7000       5500    7000   7500   6000   6500
    7500       (lost)
    8000       5500    8000   8500   7000   7500   6000   6500
    8500       (lost)
</details>

The algorithm described in the RFC requires smoltcp to treat the header's contents as a list, which is append only, remove in-place. It is not difficult, but it requires quite a bit of new code, including a new iterator on the `Assembler`.

<details>

    pub struct HolesIter<'a> {
        offset: usize,
        last_data: Option<(usize, usize)>,
        iter: DataIter<'a>
    }
    
    impl<'a> Iterator for HolesIter<'a> {
        type Item = (usize, usize);
    
        fn next(&mut self) -> Option<(usize, usize)> {
            let mut product = Some((0, 0));
            while product.filter(|(l, r)| l == r).is_some() {
                product = match (self.last_data, self.iter.next()) {
                    (Some((_, l)), Some((r, x))) => {
                        self.last_data = Some((r, x));
                        Some((l.saturating_sub(self.offset), r.saturating_sub(self.offset)))
                    },
                    (_, None) => {
                        self.last_data = None;
                        None
                    },
                    (_, _) => panic!("logic error: data iterator not fused")
                }
            }
            product.map(|(l, r)| (l + self.offset, r + self.offset))
        }
    }
</details>

Once this phase is completed, there should be significant performance gains talking to non-smoltcp TCP/IP stacks.

### Receiver Side Phase 1

This is to implement basic header handling when smoltcp is the receiver. Currently, smoltcp completely ignores any sender's header if it sends one, which is legal but not very nice.

This straightforward implementation is designed to be simple:
* Create a new `Assembler` for the remote side, based on the contents of headers our side has received.
* Send packets where the `Assembler` has a hole when it is to our advantage.
* Once we send a packet from the `Assembler`, fill the hole. (This is to handle falsely advertised windows, and avoids messing up other timeout algorithms.)

This is only an incremental improvement in practice over completely ignoring the window, but it would get the groundwork into the project. I expect there will be a lot of messy integration work, in addition to testing required.

### Receiver Side Phase 2

This final phase is the big one: generate the correct packet sequences in response to the RFC case above.

There are also some questions about timeouts will have to be explored. The RFC states that all information in the header is _advisory_, and so "over-eager senders" will eventually need to be caught, and revert back to the default "last acknowledged" part of the header if they are unsatisfied with any pattern of data generated by their assembler.

So, it's a lot of work, including a lot of testing. I hope you are still interested in the challenge!

loriopatrick, 2021-03-23:
Is anyone currently working on this or planning to pick it up?

Dirbaio, 2021-03-23:
Not as far as I know.

loriopatrick, 2021-03-24:
:+1:, getting a start on it: https://github.com/smoltcp-rs/smoltcp/issues/256

jhwgh1968, 2021-03-25:
@loripatrick I presume you meant to link to #450

loriopatrick, 2021-03-25:
Oops, yes that's the link

kasper0406, 2026-10-19:
I have been running a benchmark using a Rust-based Tailscale exit node, and measured how it performs over a noisy Wi-Fi link.

With stock smoltcp (based on what is currently on main), downloads spent 14–19% of their time stalled in 1s RTOs.

With an AI-generated patch that added bounded sender-side SACK, NewReno partial-ACK recovery and a 200ms minimum RTO, the stall time dropped to <1%, and the download speed rose from ~9.3 to ~15–16 Mbit/s. On a second, less lossy link, separating the parts suggests SACK is what carries the throughput gain (~70% over NewReno-only recovery in a small test).

Given the ongoing work here, I am curious if there is anything I can do to help get this over the line?

JPDye, 2026-10-19:
If you want to [review this](https://github.com/smoltcp-rs/smoltcp/pull/1195), that would be appreciated. And I'll follow up with the other side of SACK shortly. Haven't felt a rush on this so have been slow extracting the changes from my private fork. If  you're waiting on it, I'll speed it up.

Regarding the RTOs though, [A PR](https://github.com/smoltcp-rs/smoltcp/pull/1211) to make them configurable was merged. In the meantime, you should probably turn those down to 200ms (like Linux).